facebookresearch/habitat-lab #3: T、op-down orthographic ﬂoorplan

pbw-Berwin, 2019-03-01:
## Steps to reproduce

  1. _____
  2. _____
  3. _____

## Observed Results

  * What happened?  This could be a description, log output, etc.

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

mathfac, 2019-03-01:
Thank for the request. This functionality is in planned to be released. Multiple floors existence brings ambiguity to the task.
Worthwhile you can sample navigable points in the house and plot them. That can be considered as approximation of the floorplan.

xcyan, 2019-03-05:
@mathfac What's the timeline for releasing this feature? Thanks!

abhiskk, 2019-03-08:
I will put together an initial version for this and share here soon, this should be enough to help you in visualization.

erikwijmans, 2019-04-06:
These changes have landed: https://github.com/facebookresearch/habitat-api/blob/master/habitat/utils/visualizations/maps.py

mathfac, 2019-06-14:
Closing as functionality is already in master.

closed 2019-06-14 by mathfac