facebookresearch/habitat-lab #4: Does this include the physical engine？

Megatron2032, 2019-03-06:
how can i  integrate habitat and physical env that developed with pybullet

msavva, 2019-03-06:
Hi @Megatron2032 , thank you for your interest in the Habitat stack.  We are currently working on integration with physics engines (including bullet).  It will be implemented as additional functionality in Habitat-Sim and integrated with Habitat-API in the near future.

yashdusing, 2019-03-11:
@msavva Hello ! I wanted to ask if there's functionality to somehow stop and start habitat environments on the go. I mean, is it possible to save the context of the current environment and load it at a later time ?

msavva, 2019-03-11:
Hi @yashdusing , thank you for trying out the Habitat stack.  You can stop and start environments as desired, but we don't currently have an explicit state saving and loading mechanism in the api.  However, you can set the agent state for a particular environment, which coupled with agent state querying can be used to implement the functionality you described.

mathfac, 2019-03-15:
Closing the issue as the question is answered. Thank you!

closed 2019-03-15 by mathfac