facebookresearch/habitat-lab #5: Library: habitat_sim_bindings.so -- Supported Python Version: Unknown

LlamaSi, 2019-03-07:
## Observed Results
I have python2.7, python 3.5 and python 3.6 installed on my gpu laptop, ubuntu 16.04.
I tried to install api with python3.6, and ./build.sh as indicated in sim readme.
I tested the sim, it works fine.
Then I test on example/example.py, error occurs
Traceback (most recent call last):
  File "~/habitat-sim/habitat_sim/bindings/__init__.py", line 10, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ModuleNotFoundError: No module named 'habitat_sim._ext'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "~/habitat-sim/habitat_sim/bindings/dev_bindings.py", line 72, in <module>
    import habitat_sim_bindings
ImportError: ~/habitat-sim/build/esp/bindings/habitat_sim_bindings.so: undefined symbol: _Py_ZeroStruct

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/example.py", line 9, in <module>
    import demo_runner as dr
  File "~/habitat-sim/examples/demo_runner.py", line 15, in <module>
    import habitat_sim
  File "~/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .simulator import *
  File "~/habitat-sim/habitat_sim/simulator.py", line 7, in <module>
    from habitat_sim.bindings import *
  File "~/habitat-sim/habitat_sim/bindings/__init__.py", line 18, in <module>
    from .dev_bindings import *
  File "~/habitat-sim/habitat_sim/bindings/dev_bindings.py", line 84, in <module>
    raise ImportError(msg)
ImportError: 
Could not import habitat sim bindings
Please follow the building instructions in the README
-----------------------------------------------------------

Found the following libraries

 Library: habitat_sim_bindings.so -- Supported Python Version: Unknown

Your python interpreter is version 36
Please re-build habitat sim with this version of python

## Expected Results

I have tried to install api for 2 days. Any advice will be appreciated! Thanks!

erikwijmans, 2019-03-07:
We currently don't have a nice way to specify different versions of system python and will just generate bindings for whatever `python` is (which generally defaults to 2.7).  We highly recommend using conda environments to fix this (things should work as expected there, and `python setup.py install` will work).

If you want to stick with system python, there is a work-around for using different versions of system python here: https://github.com/facebookresearch/habitat-sim/issues/2

LlamaSi, 2019-03-07:
Thanks. Conda solved my problem.

closed 2019-03-07 by LlamaSi