facebookresearch/habitat-lab #6: Create multiagent environment

LlamaSi, 2019-03-07:
## Expected Results
My goal is to create a scene where there are other agents moving around, an ego view percept these movements. Also, I would like to keep all the images from the ego view along with the trajectory and angles of the ego camera. Is the platform compatible with this kind of task? Any potential tutorial to look at?

Thanks!

abhiskk, 2019-03-07:
@LlamaSi, this feature is currently not supported.

We definitely have multi-agent support in our roadmap but it not present right now. Although there is definitely a pseudo multiagent experimentation setup that you can do: 

1. Have multiple environments setup similar to [`train_ppo.py`](https://github.com/facebookresearch/habitat-api/blob/master/baselines/train_ppo.py).
2. Have the multiple environment be setup inside the same house.
3. In each environment you can have a single agent. You can run the multiple agents in parallel inside multiple environments but the same house (pseudo-multiagent).
4. Note that currently there is no way for the multiple agents to see each other or physically interact with each other (collisions etc).
5. Although you can setup interactions for the agents on your side (for example communication between the agents etc).

LlamaSi, 2019-03-07:
Thanks a lot for the answer!
May I ask if a simpler case with one agent and multiple dynamic objects is possible? Like some mesh moving around inside the scene. Can the agent see those moving objects?

abhiskk, 2019-03-08:
@LlamaSi at the moment it is not possible. A simple feature of moving mesh would go hand in hand with a better multiagent support. But this feature is not in place right now.

closed 2019-03-08 by LlamaSi

xiaotaw, 2019-11-25:
Looking forward to multi-agent support too. [smile]

srama2512, 2020-01-06:
@abhiskk - what would be the timeframe for enabling multi-agent support in Habitat? Can it be expected in the near future? Also, are there any plans to have a standard implementation of pseudo multiagent?

mathfac, 2020-01-06:
Hi @srama2512! We will have team planning in near future and let you know.

smorad, 2021-07-07:
Has there been any update on this? I see multi agent support exists many places in the codebase, but there are no guides on how to add additional agents. Adding `AGENT_1, AGENT_2, ...` to the habitat config doesn't work.

JiayunjieJYJ, 2021-09-14:
> Has there been any update on this? I see multi agent support exists many places in the codebase, but there are no guides on how to add additional agents. Adding `AGENT_1, AGENT_2, ...` to the habitat config doesn't work.

> Has there been any update on this? I see multi agent support exists many places in the codebase, but there are no guides on how to add additional agents. Adding `AGENT_1, AGENT_2, ...` to the habitat config doesn't work.

I have the same question, do you know how to create multi agent environment in habitat?