facebookresearch/habitat-lab #7: Choosing what scene is loaded in Simulator

a7b23, 2019-03-18:
Hi, line [100](https://github.com/facebookresearch/habitat-api/blob/master/baselines/train_ppo.py#L100) mentions that config_env.SIMULATOR.SCENE = dataset.episodes[0].scene_id . Does this mean that the agent is always trained on just the first scene-id present in the val.json.
Also, the train.json file is empty, so how are the different scenes loaded?

mathfac, 2019-03-18:
Hi @a7b23, that means that first scene that Simulator will load will be the one specified in the first episode of a dataset.
Regarding `train.json` all episodes are loaded from subfolder `content` that contains scene based episodes.

mathfac, 2019-04-03:
Feel free to re-open the issue, if you still have questions.

takuma-ynd, 2019-05-03:
I understand that when we use `train.json`, all episodes are loaded from subfolder `content`, but how can we switch those loaded environments manually?

When I read [train.ppo](https://github.com/facebookresearch/habitat-api/blob/master/baselines/train_ppo.py), I got confused because I couldn't find any code that explicitly switches environments.

In addition, I could not find anything for switching environments in [Env](https://github.com/facebookresearch/habitat-api/blob/1f0ef53ceb63f5532df3eef53d8eda163dc354dd/habitat/core/env.py#L22) class, either.

Could you clarify this point?
Thanks!

abhiskk, 2019-05-03:
@takuma-ynd if you want to switch to loading these environments manually you can modify the logic here: https://github.com/facebookresearch/habitat-api/blob/master/habitat/core/env.py#L178-L184 . On a reset call the episode index is advanced and the `reconfigure` in line 185 then loads the house from new episode into the simulator. 

I think you might have already identified this but stating it to be clear, it is important for training the PPO that the episodes are randomly sampled, but there is a trade-off here, if you randomly sample the episode it is likely that each successive episode will be in a different environment, this is not good in terms of speed as this would mean that there would be loading and dumping of graphical assets in the simulator which is usually a slow step, a way around this is if you shuffle the episodes such that consecutive `shuffle_interval` episodes happen inside the same house, in our experiments we used a `shuffle_interval` value of 500:

```python
def shuffle_episodes(self, shuffle_interval):
        ranges = np.arange(0, len(self.episodes), shuffle_interval)
        np.random.shuffle(ranges)
        new_episodes = []
        for r in ranges:
            new_episodes += self.episodes[r:r + shuffle_interval]
        self.episodes = new_episodes
```

This method would be present inside the [PointnavDatasetV1](https://github.com/facebookresearch/habitat-api/blob/master/habitat/datasets/pointnav/pointnav_dataset.py#L25) class and the method would be called just after the dataset is created (after this [line](https://github.com/facebookresearch/habitat-api/blob/master/baselines/train_ppo.py#L97))

Ideally this method should have been part of the original codebase which was opensourced, if you do end up using this or any other modification of shuffling can you go ahead and open a PR for this.

Let me know if you have any other questions. Thanks 🙂

stbook, 2019-10-13:
So how to switch scene exactly? I'm still confused after reading the response.

erikwijmans, 2019-10-15:
This is what initiates a scene switch: https://github.com/facebookresearch/habitat-api/blob/master/habitat/core/dataset.py#L249 -- the environment then just tells the simulator what scene to load for a given episode.

We currently don't have a way to force a scene switch with a method call.

smorad, 2020-03-30:
Just to update this, there now appears to be a way to force a switch, present in the episode iterator's `__next__` method

https://github.com/facebookresearch/habitat-api/blob/master/habitat/core/dataset.py#L391

rpartsey, 2022-08-31:
The example how the scene can be switched in the PointNav task (you may do the scene switch for the other tasks by following the same logic): 
```
from habitat.datasets.pointnav.pointnav_generator import generate_pointnav_episode
from habitat.tasks.nav.nav import merge_sim_episode_config

# 1. switch the scene to new_scene
sim_config.defrost()
sim_config.SCENE = new_scene
sim_config.freeze()
sim.reconfigure(sim_config)

# 2. generate the episode for new_scene
episode_gen = generate_pointnav_episode(
    sim=self.sim,
    is_gen_shortest_path=False
)
current_episode = next(episode_gen)

sim_config.defrost()
sim_config = merge_sim_episode_config(
    sim_config,
    episode
)
sim_config.freeze()
sim.reconfigure(sim_config)

# 3. reset the sim
obs = self.sim.reset()

# 4. do what you want
```

For more detailed implementation you may refer to this [code](https://github.com/rpartsey/pointgoal-navigation/blob/80d3375d1bb4898fe04f9379ee38059049be4263/odometry/dataset/dataset.py#L290).

Feel free to re-open the issue, if you still have questions.

closed 2022-08-31 by rpartsey