facebookresearch/habitat-lab #9: SLAM baselines

Toraudonn, 2019-04-02:
Thanks for your hard work!
Couple of questions regarding the SLAM baseline:

- When will it be released?
- If I wanted to make my own custom SLAM for Habitat, where should I edit?

abhiskk, 2019-04-02:
@Toraudonn we do have a working SLAM baseline but the code quality is not great, I can work to release the dirty version we have soon. 

If you wanted to create your own SLAM method for Habitat I think you would need access to only the `env.step` and `env.reset` methods. You can refer to [`train_ppo.py`](https://github.com/facebookresearch/habitat-api/blob/master/baselines/train_ppo.py) which trains an agent for navigation using PPO (RL). The dynamics should stay the same for SLAM.

ducha-aiki, 2019-04-05:
@Toraudonn https://github.com/facebookresearch/habitat-api/pull/39

Toraudonn, 2019-04-17:
@abhiskk , @ducha-aiki Thanks!

closed 2019-04-17 by Toraudonn