facebookresearch/habitat-lab #15: Overhead camera view with an actual agent

mathfac, 2019-04-03:
author @dhruvbatra:  [I believe this feature should be implemented in habitat-api not habitat-sim but please correct me if I'm mistaken]

Feature request: we should add an overhead camera sensor so we can see the agent move around like a player in a game.

See for example:
https://raw.githubusercontent.com/StanfordVL/GibsonEnv/master/misc/husky_camera.png
https://raw.githubusercontent.com/Unity-Technologies/obstacle-tower-env/master/banner.png

This could be an additional "sensor" that is attached to the agent, but the observations from this sensor are not used by the agent (the agent only uses ego-centric camera).

mathfac, 2022-10-27:
@aclegg3 do we have this functionality already in master? Thanks.

Skylion007, 2022-10-27:
Not quite. The main thing we need is to automatically compute the appropriate camera pose based on the agent's bbox.