facebookresearch/habitat-lab #38: Sensor configuration and

Jiankai-Sun, 2019-04-05:
Firstly, Thank you for your great work!
## Questions

  1. In [configs/tasks/pointnav_gibson.yaml line 5](https://github.com/facebookresearch/habitat-api/blob/master/configs/tasks/pointnav_gibson.yaml#L5) and [configs/tasks/pointnav_mp3d.yaml line 5](https://github.com/facebookresearch/habitat-api/blob/master/configs/tasks/pointnav_mp3d.yaml#L5), SENSORS for SIMULATOR is set as ['RGB_SENSOR']. However, in the observations returned by the simulator, there are still 'rgb', 'depth' and 'goal'. How can we specify the sensors we want in observations correctly?

mathfac, 2019-04-05:
Hi @Jiankai-Sun,
Firstly, thank you for giving Habitat a try.

How exactly are you running Habitat?

Usually, a config is explicitly specified as argument to ‘get_config’ method.
If you running Habitat inside the Habitat challenge local testing then a config is provided differently based on a track.

Jiankai-Sun, 2019-04-05:
Thank you for your reply @mathfac !

I run Habitat as 
```
python -u train_ppo.py \
    --use-gae \
    --sim-gpu-id 0 \
    --pth-gpu-id 1 \
    --lr 2.5e-4 \
    --clip-param 0.1 \
    --value-loss-coef 0.5 \
    --num-processes 4 \
    --num-steps 128 \
    --num-mini-batch 4 \
    --num-updates 100000 \
    --use-linear-lr-decay \
    --use-linear-clip-decay \
    --entropy-coef 0.01 \
    --log-interval 5 \
    --checkpoint-folder "data/checkpoints_gibson_4" \
    --checkpoint-interval 50 \
    --task-config "tasks/pointnav_gibson.yaml" \
```

I use the provided config file "tasks/pointnav_gibson.yaml".

mathfac, 2019-04-05:
There is an argument for `train_ppo.py` `--sensors` and it's default value is defined here: https://github.com/facebookresearch/habitat-api/blob/master/baselines/rl/ppo/utils.py#L407. It was introduced make changes of modalities easier for training.
2. Yes, you can count on that.

Jiankai-Sun, 2019-04-05:
Thank you! @mathfac 

Currently, I can only put the habitat-sim on 1 GPU defined by `--sim-gpu-id 0` and pytorch training on 1 GPU defined by `--pth-gpu-id 1`. 

My question is: can habitat simulator support multi-GPU? i.e., `--sim-gpu-id 0,1,2,3`, `--pth-gpu-id 4,5,6,8`. I have tried the above settings but it seems that Habitat doesn't support such settings. The reason is [config_env.SIMULATOR.HABITAT_SIM_V0.GPU_DEVICE_ID](https://github.com/facebookresearch/habitat-api/blob/master/baselines/train_ppo.py#L135) doesn't support multi-GPU simulator.

Jiankai-Sun, 2019-04-05:
Another question is: Could you please give the direction defined in Cartesian coordinate for Habitat? (I mean, How does Habitat define the positive direction of the Cartesian coordinate system for each scene?). 

Thank you!

erikwijmans, 2019-04-05:
A single instance of the simulator can only utilize 1 GPU at a time, however, it is able to use any GPU when built in headless mode (using the `--headless` flag, this is only supported with nvidia GPUs on linux).

Currently the baseline code isn't setup to use multiple GPUs, but it shouldn't be hard to modify.  For the simulator, you can spawn different instances on different GPUs by modifying the loop here: https://github.com/facebookresearch/habitat-api/blob/master/baselines/evaluate_ppo.py#L49 to change the GPU id based on the rank of the process.  For pytorch, proceed as normal.

The coordinate system in habitat-sim is -z forward, +y up.

erikwijmans, 2019-04-05:
> @Jiankai-Sun  Are all the instances in the gibson/mp3d train/val/test set reachable within 500 steps?

We didn't explicitly control for this when generating the datasets, and for mp3d it is feasible that an episode wouldn't be achievable in 500 steps (for Gibson, this seems incredibly unlikely if not impossible since the houses are so much smaller).  I have yet to see this actually be the case however.

Jiankai-Sun, 2019-04-06:
> The coordinate system in habitat-sim is -z forward, +y up.

Thank you for your reply!

Could you please give a sketch map to illustrate the "The coordinate system in habitat-sim is -z forward, +y up"?

Thank you!

mathfac, 2019-04-06:
To support @erikwijmans's answer, to create separate simulator on different GPU for **PPO training** next line should be changed during config creation for child processes https://github.com/facebookresearch/habitat-api/blob/master/baselines/train_ppo.py#L135:
```config_env.SIMULATOR.HABITAT_SIM_V0.GPU_DEVICE_ID = int(args.sim_gpu_id.split(",")[i % len(sim_gpu_id.split()))```.

erikwijmans, 2019-04-06:
I am not entirely certain how to make a sketch to illustrate the coordinate system, but I will try to restate it in a different way:

The coordinate system is a normal Cartesian coordinate system, where -z is the forward direction.  This is to say that a camera with no rotation is looking along the negative of the z-axis and changing the camera's height moves it along the y-axis.

The floors in the buildings are in the x-z plane, and the agent travels in the x-z plane (assuming it isn't going up stairs).   When the agent has no rotation, the forward action move the agent along the negative of the z-axis.

Jiankai-Sun, 2019-04-07:
Thank you for your quick reply!

closed 2019-04-07 by Jiankai-Sun