facebookresearch/habitat-lab #57: goal distance is computed inconsistently for PointGoalSensor and SPL measure

karkuspeter, 2019-04-15:
For the navigation task the computation of goal distance is inconsistent for the PointGoal sensor [here](https://github.com/facebookresearch/habitat-api/blob/9a9927cc9ae852bb3bb0376f7ed6f9ece9def470/habitat/tasks/nav/nav_task.py#L197-L201) and the SPL measure [here](https://github.com/facebookresearch/habitat-api/blob/9a9927cc9ae852bb3bb0376f7ed6f9ece9def470/habitat/tasks/nav/nav_task.py#L317-L319). The former seems to be a 2D distance while the latter seems to be 3D distance.

As a result, in some cases the agent is within desired distance to the goal according to the PointGoal sensor, but not according to the SPL code.

mathfac, 2019-04-15:
Hi @kakuspeter,
PointGoal sensor provides Euclidian distance, while SPL uses geodesic shortest path distance. That’s not enough to reach a goal that is around the wall.

Hope that answered your question.

erikwijmans, 2019-04-15:
To add on to what @mathfac said, for this year of the habitat challenge, we have decided that all episodes of PointNav start and end on the same floor, thus the 2D vector to the goal is enough information to successfully complete the episode.  We have seen that blind agents have difficulty with realizing that they have gone up/down stairs (because how would they know), but models with vision seem to be able to overcome this.

karkuspeter, 2019-04-16:
Thanks for the quick reply!

The issue is not only with walls or moving between floors. In the example below the y coordinate of the goal is defined ~10cm higher to what agent achieves. The Euclidian distance is less than the threshold (0.2), while the geodesic distance is larger than the threshold. The agent has no way to tell that the goal is not reached.

## Steps to reproduce

  1. change line 27 in configs/tasks/pointnav_gibson.yaml from `SPLIT: train` to `SPLIT: val`
  2. run code below

## Observed Results
Position:
[-11.068585     0.24531515  -0.13574535]
Goal:
[-11.20691872   0.35344285  -0.02380179]
Pointgoal distance:
0.17795356
Geodesic distance:
0.20822839438915253
Eucledian dist (3D):
0.20822839774681173
Eucledian dist (2D):
0.1779535514904323
SPL:
0.0

## Expected Results
SPL > 0.0

## Relevant Code

```
import habitat, os, numpy as np

env = habitat.Env(
    config=habitat.get_config(config_file="tasks/pointnav_gibson.yaml")
)

for _ in range(32):
    env.reset()

actions = [1, 1, 1, 0, 0, 2, 0, 0, 0, 0, 1, 1, 1, 0, 2, 2, 0, 0, 1, 1, 0, 2, 2, 2, 2, 0, 0, 0, 2, 2, 2, 2, 2, 2, 2, 0, 0, 3]

for act in actions:
    obs = env.step(act)

pos = env.sim.get_agent_state().position
goal = np.array(env.current_episode.goals[0].position)
print ("Position:")
print (pos)
print ("Goal:")
print (goal)
print ("Pointgoal distance:")
print (obs['pointgoal'][0])
print ("Geodesic distance:")
print (env.sim.geodesic_distance(pos, goal))
print ("Eucledian dist (3D): ")
print (np.linalg.norm(pos - goal))
print ("Eucledian dist (2D): ")
print (np.linalg.norm(pos[((0,2), )] - goal[((0,2),)]))
print ("SPL:")
print (env.task.measurements.get_metrics()['spl'])
```

mathfac, 2019-04-17:
Thank you for flagging this.
Yes, agent has no way to tell that goal is reached as it has no access to geodesic shortest path distance anyway. While model can still learn right criteria during training.

For future we will use 3D Euclidian distance for PointGoal.

abhiskk, 2019-04-17:
Thanks for digging more into the issue @karkuspeter , the problem that you point out is valid, going ahead we will switch to using a 3D pointgoal vector as input. If you want you can use it for your own experiments by changing the `GOAL_FORMAT` from `CARTESIAN` to `POLAR`. But we won't be making any change on this for the challenge setup, this is to keep things simple for now and stick to our initial setting (same as the one we used in our paper (https://arxiv.org/abs/1904.01201).

Please do keep this issue open so that we can track it and make a change later.

karkuspeter, 2019-04-18:
Thanks.
@abhiskk do you mean changing GOAL_FORMAT to POLAR in the config file? It is actually set to POLAR by default in pointnav_gibson.yaml.

abhiskk, 2019-04-18:
Sorry my mistake, I meant changing `GOAL_FORMAT` from `POLAR` to `CARTESIAN`

mathfac, 2019-06-15:
Looks like this issue is solved, feel free to reopen if needed.

closed 2019-06-15 by mathfac