facebookresearch/habitat-lab #59: Some objects of some scenes in the MatterPort3D dataset are void and not accessible

eric-xw, 2019-04-16:
I tried to iterate the scenes in the MatterPort3D dataset and obtain all the object information. I did something similar as follows:
```
scene = env.sim.semantic_annotations()
for obj in scene.objects:
            print(obj)
```

It turned out that some objects are void and cannot be accessible. So I encounted the error outout
```
<habitat_sim_bindings.SemanticObject object at 0x7f2bd02260d8>
<habitat_sim_bindings.SemanticObject object at 0x7f2bd0226110>
Segmentation fault (core dumped)
```

The scene id for this error is `ur6pFq6Qu1A`, which is from the train split. But there are also some other problematic scenes in all splits (train, val, test), which you need also to check. 

Thanks,
Xin

erikwijmans, 2019-04-16:
I am unable to reproduce this, the following works fine for me:

```
import habitat_sim

cfg = habitat_sim.SimulatorConfiguration()
cfg.agents = [habitat_sim.AgentConfiguration()]
cfg.scene.id = "data/scene_datasets/mp3d/ur6pFq6Qu1A/ur6pFq6Qu1A.glb"
sim = habitat_sim.Simulator(cfg)

for obj in sim.semantic_scene.objects:
    print(obj)
```

(I tried to reproduce directly on the simulator side instead of through the API as that is where the bug would be for this issue).

eric-xw, 2019-04-16:
the error is there every time when iterating that specific scene. I am using the latest Habitat simulator code.
```
I0416 14:30:23.078264 24426 Simulator.cpp:151] Loading navmesh from data/scene_datasets/mp3d/ur6pFq6Qu1A/ur6pFq6Qu1A.navmesh
I0416 14:30:23.078892 24426 Simulator.cpp:153] Loaded.
I0416 14:30:23.078929 24426 Simulator.cpp:168] Loading house from data/scene_datasets/mp3d/ur6pFq6Qu1A/ur6pFq6Qu1A.house
I0416 14:30:23.157354 24426 Simulator.cpp:175] Loading semantic mesh data/scene_datasets/mp3d/ur6pFq6Qu1A/ur6pFq6Qu1A_semantic.ply
I0416 14:30:23.157397 24426 ResourceManager.cpp:198] loading MP3D instance mesh data: data/scene_datasets/mp3d/ur6pFq6Qu1A/ur6pFq6Qu1A_semantic.ply
I0416 14:30:25.044412 24426 ResourceManager.cpp:207] Building SceneGraph
I0416 14:30:25.044942 24426 Simulator.cpp:187] Loaded.
I0416 14:30:25.045024 24426 Simulator.cpp:217] Reset agent i=0 position=0.402639 0.126979 -22.3185 rotation=00000000 0.989856 00000000 0.142074
I0416 14:30:25.047144 24426 Simulator.cpp:217] Reset agent i=0 position=-14.3553 0.126979 0-15.651 rotation=00000000 0.890628 00000000 0.454732
--------------------------------------------------EPISODE 69--------------------------------------------------
<habitat_sim_bindings.SemanticScene object at 0x7f53df554f48>
<habitat_sim_bindings.SemanticObject object at 0x7f53df6943e8>
<habitat_sim_bindings.SemanticObject object at 0x7f53e2c42650>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e93e8>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9378>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9490>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9458>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9340>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9420>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e94c8>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9500>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9538>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9570>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e95a8>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e95e0>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9618>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9650>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9688>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e96c0>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e96f8>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9730>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9768>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e97a0>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e97d8>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9810>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9848>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9880>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e98b8>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e98f0>
<habitat_sim_bindings.SemanticObject object at 0x7f53e33e9928>
Segmentation fault (core dumped)
```

erikwijmans, 2019-04-16:
Are you doing anything with the objects other than printing them? i.e. calling a method/accessing a property?

eric-xw, 2019-04-16:
Yes. I also called `obj.id`. Either printing it or accessing the property caused the same issue.

erikwijmans, 2019-04-16:
Ah, looks like we aren't properly handling objects that don't have a corresponding region.

eric-xw, 2019-04-17:
I pulled the new changes of Habitat-sim and rebuilt it. It seems the object can be accessible now.

BTW, the code is still buggy (other attributes like `category` other than `id` cannot be printed):
```
import habitat_sim

cfg = habitat_sim.SimulatorConfiguration()
cfg.agents = [habitat_sim.AgentConfiguration()]
cfg.scene.id = "data/scene_datasets/mp3d/ur6pFq6Qu1A/ur6pFq6Qu1A.glb"
sim = habitat_sim.Simulator(cfg)

scene = sim.semantic_scene
for obj in scene.objects:
    print(obj.id)
    print(obj.category.name())
sim.close()
```
The error message is
```
I0417 14:52:42.191834 60709 Simulator.cpp:217] Reset agent i=0 position=-3.39043 0.135602 0-25.444 rotation=00000000 0.596784 00000000 0.802402
0_0_0
double free or corruption (out)
Aborted (core dumped)```

closed 2019-04-21 by eric-xw

BryanBo-Cao, 2019-11-20:
May I know how to get the ```*. glb``` file from the downloaded Matterport3D dataset? I've downloaded the entire dataset, but what I see are all the ```*.zip```:
e.g.
```
dataset/Matterport3D/v1/scans/ur6pFq6Qu1A$ ls
cameras.zip              image_overlap_data.zip            matterport_camera_poses.zip  matterport_depth_images.zip  matterport_mesh.zip           poisson_meshes.zip        sens.zip                           undistorted_color_images.zip  undistorted_normal_images.zip
house_segmentations.zip  matterport_camera_intrinsics.zip  matterport_color_images.zip  matterport_hdr_images.zip    matterport_skybox_images.zip  region_segmentations.zip  undistorted_camera_parameters.zip  undistorted_depth_images.zip
```
As far as I know the code refers to ```*.glb``` file.

May I know if I missed something? Any help would be appreciated!

msavva, 2019-11-20:
@BryanBo-Cao you will need to follow the instruction at https://github.com/facebookresearch/habitat-api#matterport3d 
Specifically you need to call `python download_mp.py --task habitat -o data/scene_datasets/mp3d/` to download the glb format files (you do not need the rest of the Matterport3D dataset).

BryanBo-Cao, 2019-11-22:
@msavva Thanks for your reply. I found the ```*glb``` files in the ```*.zip``` file.

GaganBhat, 2022-02-04:
@msavva Where is `download_mp.py` located? It's not in this repository, nor in any of Matterport3D habitat website.

msavva, 2022-02-04:
@GaganBhat follow the instructions on the Matterport3D website to request access to the dataset.  You will get the script when you are granted access.

GaganBhat, 2022-02-04:
@msavva I got approved, looks like they have changed how they distribute the data, cannot find that script anywhere. 

Here's the link to the public repo they shared with me after approval -- https://github.com/matterport/habitat-matterport-3dresearch

msavva, 2022-02-04:
You are referring to the HM3D dataset (see https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md for information on HM3D).  Note that HM3D is different than the Matteport3D dataset (which uses the `download_mp.py` script).

GaganBhat, 2022-02-04:
@msavva Sorry for the confusion! Yes, the MP3D dataset does in fact come with the `download_mp.py` script. Got it.

ylqi, 2022-11-23:
> @msavva Thanks for your reply. I found the `*glb` files in the `*.zip` file.

Hi, I cannot find any `*.glb` files in those `*.zip` files. Those `*.zip` files include files such as `*.png`, `*.sens`, `*.ply`, `*.json` or `*.cam`. May I know which `*.zip` file includes the `*.glb` files?