facebookresearch/habitat-lab #60: A shortest path following example of using VectorEnv?

eric-xw, 2019-04-16:
Hi,

Can you provide an example of using VectorEnv for shortest path following? similar to this one (https://github.com/facebookresearch/habitat-api/blob/master/examples/shortest_path_follower_example.py) but with multiple processes. 

Thanks,
Xin

mathfac, 2019-06-14:
We don't have such tutorial yet.

gianscarpe, 2021-06-30:
Hi! Is this still of any interests?

rpartsey, 2022-08-31:
Hi @eric-xw @gianscarpe 

If you are still interested in using the VectorEnv for shortest path following, this should be pretty easy to implement.

You may define a thin wrapper on top of the Env class (let's say, SPFEnv) that should have attributes env (Env instance) and spf ()

rpartsey, 2023-01-26:
Closing due to inactivity. Feel free to re-open the issue, if you still have questions.

closed 2023-01-26 by rpartsey