facebookresearch/habitat-lab #61: How to render the environment?

koriavinash1, 2019-04-17:
Is it possible to render environment to behavior of the trained agent?

mathfac, 2019-04-17:
Hi @koriavinash1,
Yes, if you enable RGB sensor you can save RGB observations during navigation as video using code from this PR: https://github.com/facebookresearch/habitat-api/pull/48/files.

rpartsey, 2022-08-31:
Yet another alternative is to enable a `TOP_DOWN_MAP` sensor (as it was done in the [shortest_path_follower_example.py](https://github.com/facebookresearch/habitat-lab/blob/main/examples/shortest_path_follower_example.py#L48)). 
Visualisation examples are demonstrated in the [visualization_examples.py](https://github.com/facebookresearch/habitat-lab/blob/dcfced6259ae3f7d1068db0c20e3f82658e8dd60/examples/visualization_examples.py#L101) too.

Closing the issue because it has not had recent activity.

closed 2022-08-31 by rpartsey