facebookresearch/habitat-lab #63: How to initialize environment outside of main repo?

bradleyemi, 2019-04-19:
## Steps to reproduce

  1. I cloned `examples/example.py` outside of the repository (i.e. in a different project repo) and attempted to run it.

## Observed Results

  I got `No such file or directory: 'data/datasets/pointnav/gibson/v1/train/train.json.gz'`. because the data path in the config is a relative path. When I changed DATA_PATH in `configs/tasks/pointnav_gibson.yaml` I got 

## Expected Results

 I expected the script to run as normal.

How do I make an environment from config from outside the main repo? We'd like to integrate Habitat into our existing project as an environment and are having extreme trouble just initializing an environment from outside `habitat-api`. Thank you so much for the help!

bradleyemi, 2019-04-19:
After changing the DATA_PATH to the absolute path of the config, I get:
```
~/svl/habitat/feature_selector/habitat-sim/habitat_sim/simulator.py in _config_backend(self, config)
     52     def _config_backend(self, config: Configuration):
     53         if self._sim is None:
---> 54             self._sim = hsim.SimulatorBackend(config.sim_cfg)
     55         else:
     56             self._sim.reconfigure(config.sim_cfg)

ValueError: Cannot load: data/scene_datasets/gibson/Adrian.glb
```

abhiskk, 2019-04-19:
You can create a new train pointnav dataset setup, the new setup will be mostly identical to the current setup, you can use the current [`pointnav_gibson.zip`](https://github.com/facebookresearch/habitat-api/#task-datasets) as the starting point. The train dataset for Gibson is split over different houses, so the folder `pointnav/gibson/v1/train/content` contains files like `Adrian.json.gz`, `Albertville.json.gz` which contains the different episodes inside the respective houses. Each `episode` contains a `scene_id` which is of the type `data/scene_datasets/gibson/Adrian.glb`. You will have to modify this value to the location where you have stored the glbs. 

Let me know if you have anymore questions and thanks for trying this out, I do recognize that this kind of setup is slightly convoluted and we are trying to simplify this setup.

erikwijmans, 2019-04-19:
The "just make it work" fix is to do `ln -s /path/to/habitat-api/data` while in the folder for the different project's repo

mathfac, 2019-04-30:
@bradleyemi with https://github.com/facebookresearch/habitat-api/pull/77 we have parameter `config.DATASET.SCENES_DIR` to control the location of scenes dataset and get rid of `data` symlink requirement. Let us know if you have any questions regarding that and feel free to reopen the issue.

closed 2019-04-30 by mathfac