facebookresearch/habitat-lab #64: openai baseline api change

danielgordon10, 2019-04-19:
dict_space was renamed dict and now it won't import with current versions of openai baselines. 
https://github.com/openai/gym/commit/f5d571a16d18d3a64d56819c76e803a89149397c#diff-b6bcac3b96f9b9aa1c6be76eaa754d20

Either we should specify exactly which commit to install, or the interface should be updated.

mathfac, 2019-04-26:
@danielgordon10  do you advice update gym version in `requirements.txt` and update the code accordingly?

danielgordon10, 2019-04-27:
Actually, with the openai gym pinned to a specific version, this may have been my fault based on install order. I guess it's fine as is, but it could be updated to match also. Since it's kind of a small dependency, I wouldn't worry about it. I'll close the issue.

closed 2019-04-27 by danielgordon10