facebookresearch/habitat-lab #72: Dockerfile for SLAM agent

kudo1026, 2019-04-24:
https://github.com/facebookresearch/habitat-api/blob/8c5329ce9f777395b45c2daac135377e629ec148/Dockerfile#L49

Is it possible to update the Dockerfile to have the SLAM agent running? The install script "install_deps.sh" doesn't seem to work for now.

Big thanks in advance!

kudo1026, 2019-04-24:
Error information by running "install_deps.sh"
```
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++11.
-- Found PythonLibs: /opt/conda/envs/habitat/lib/libpython3.6m.so (found suitable version "3.6.8", minimum required is "3.6")
-- Found NumPy: /opt/conda/envs/habitat/lib/python3.6/site-packages/numpy/core/include (found version "1.16.3")
CMake Warning at /opt/cmake/share/cmake-3.13/Modules/FindBoost.cmake:1775 (message):
  No header defined for python-py35; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:33 (find_package)

CMake Error at /opt/cmake/share/cmake-3.13/Modules/FindBoost.cmake:2100 (message):
  Unable to find the requested Boost libraries.

  Unable to find the Boost header files.  Please set BOOST_ROOT to the root
  directory containing Boost or BOOST_INCLUDEDIR to the directory containing
  Boost's headers.
Call Stack (most recent call first):
  CMakeLists.txt:33 (find_package)

-- Found OpenCV: /opt/conda/envs/habitat (found version "3.4.2")
-- Found ORB_SLAM2: /habitat-api/baselines/slambased/3rdparty/ORBSLAM2_installed/lib/libORB_SLAM2.so
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Boost_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /habitat-api/baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings
   used as include directory in directory /habitat-api/baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings
   used as include directory in directory /habitat-api/baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings
   used as include directory in directory /habitat-api/baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings
   used as include directory in directory /habitat-api/baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings
   used as include directory in directory /habitat-api/baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings
   used as include directory in directory /habitat-api/baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings

-- Configuring incomplete, errors occurred!
See also "/habitat-api/baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings/build/CMakeFiles/CMakeOutput.log".
./install_deps.sh: line 60: /habitat-api/baselines/slambased/3rdparty/eigen3_installed/include/eigen3: Is a directory
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
```

mathfac, 2019-04-25:
Hi @kudo1026,
Thanks for reaching out. Unfortunately, we don't have Dockerfile for SLAM agents.
Regarding the error, looks like boost library is missing. Can you try run next commands and rerun `install_deps.sh`
```
source activate HandcraftedAgents
conda install -c conda-forge boost-cpp -y
```
cc @ducha-aiki .

rpartsey, 2022-08-31:
Closing the issue because it has not had recent activity.

closed 2022-08-31 by rpartsey