facebookresearch/habitat-lab #83: enforce determinism in evaluate_ppo.py?

ChengshuLi, 2019-05-11:
https://github.com/facebookresearch/habitat-api/blob/5679a5dc26d5097eb14658eff91a800e18763fe6/habitat_baselines/evaluate_ppo.py#L118-L123

Why is the flag deterministic set to be False here? Shouldn't it be True for evaluation?

Thanks a lot!

ChengshuLi, 2019-05-11:
Also, during evaluation, should we also consider calling self.eval() for `Net`? My understanding is that if `Net` contains dropout or batch normalization, we need to set model.eval() before evaluation. Any help will be greatly appreciated. Thanks again! 

https://github.com/facebookresearch/habitat-api/blob/ac166fa8d706a150666f37a3830505c5fd5c0d4c/habitat_baselines/rl/ppo/policy.py#L62-L85

erikwijmans, 2019-05-11:
We use `deterministic=False` as we found that sampling actions led to better performance than argmaxing.  In the example agent code, we do fix the random seed so that the code is still deterministic (although that seems to have been omitted from the eval code).

erikwijmans, 2019-05-11:
Yes, if your model uses batch norm or dropout, you should call `model.eval()` before evaluation.  This should probably also be done before sampling actions in the rollout.

ChengshuLi, 2019-05-11:
Thanks for your prompt response. :)

closed 2019-05-11 by ChengshuLi