facebookresearch/habitat-lab #87: [ORBSLAM2 Agent bug] estimatedGoalPos outside map bounds

mcimpoi, 2019-05-17:
Thank you for the great work and releasing the simulator and baselines.

I am trying to run the SLAM baseline on the MP3D val split, which runs for the first 19 episodes, and crashes on the 20th.

## Steps to reproduce

  1. create point_nav_mp3d_val.yaml from point_nav_mp3d.yaml provided, with:
SENSORS: ['RGB_SENSOR', 'DEPTH_SENSOR'] and SPLIT: val
  2. python habitat_baselines/agents/slam_agents.py --task-config configs/tasks/pointnav_mp3d_val.yaml

## Observed Results

  * when evaluating episode 20 (scene: 2azQ1b91cZZ.glb, start_position: [16.921436309814453, 0.12711000442504883, 9.845909118652344], goal_position: [4.39768648147583, 0.12711000442504883, -7.979780673980713] ), the 
   IndexError: index 418 is out of bounds for dimension 3 with size 400
 (caused by: self.estimatedGoalPos2D = tensor([[201., 418.]], which is then used to set ones in the goal_map which is of size [1,1,400,400]).

## Expected:
 - handle value out of bounds / not get an estimated position outside the bounds?

## Relevant Code
habitat-api/habitat_baselines/agents/slam_agents.py(L472) -- plan_path()

Any hints how to fix this?
Many thanks in advance.

mathfac, 2019-05-23:
Thank you for reaching out. cc @ducha-aiki as author of the code and @abhiskk successfully run slam baseline on val split.

ducha-aiki, 2019-05-24:
Hi,
This is because map size for planner is limited for limiting GPU memory usage. I tried to use constant which is enough to for maps tested, yet not too big. 
You can increase it here:

https://github.com/facebookresearch/habitat-api/blob/60653f7b480db604fe05786c5909a308beea9f5e/habitat_baselines/config/default.py#L40

mcimpoi, 2019-05-27:
Thank you! 45 seems to work fine.

closed 2019-05-27 by mcimpoi