facebookresearch/habitat-lab #95: Missing :- handcrafted-agent-example.ipynb

kowshikthopalli, 2019-05-31:
Hi,
Thanks for this wonderful repo.
In slam based agents folder in habitat baselines , it has been stated that there is a notebook titled 'handcrafted-agent-example.ipynb ' .
I could not find the same. Can you please point to it?

mathfac, 2019-06-14:
Maybe @ducha-aiki can share it with you.

ducha-aiki, 2019-06-16:
Hi,
It was lost when adopting to version changes, but I can recreate and upload it next week.

Tiga002, 2019-11-25:
@ducha-aiki  
Hi ! Thanks for your great effort ! 
However, I still couldn't find the handcrafted-agent-example.ipynb in the master branch. 
Would you like to upload it again ?

keita-rgit, 2020-02-12:
Hi,
Thank you for making excellent work.
I want to try the SLAM in Habitat too, so please upload "../handcrafted-agent-example.ipynb".
I tried it by following the instructions below, but I didn't know how to run the slambased.
[https://github.com/facebookresearch/habitat-api/tree/master/habitat_baselines/slambased](url)

ducha-aiki, 2020-02-13:
Hi,

I will upload it as soon as be able to install habitat itself :)

https://github.com/facebookresearch/habitat-sim/issues/491

manikamakam, 2020-12-12:
Hello, 

I cannot find the "handcrafted-agent-example.ipynb". Can someone point me towards it or any other resource which explains how to use their 'slambased' baseline? Thanks in advance.

morpheus1820, 2022-04-10:
Hi! Has the example notebook "handcrafted-agent-example.ipynb" been made available yet?

dhruvbatra, 2022-04-10:
I think it's safe to assume that at this point that we no longer support any SLAM-based agent with the current version of the code.

closed 2022-04-10 by dhruvbatra