facebookresearch/habitat-lab #96: slambased/install deps.sh failing

kowshikthopalli, 2019-05-31:
## Steps to reproduce

  1. On a server with no sudo access, run install_deps.sh after uncommenting line 8
 
## Observed Results

  * `
(HandcraftedAgents) kthopalli@ubuntu:~/habitat-api/habitat_baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings/build$ make install
Build type: Release
-- Using flag -std=c++11.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   python-py35
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kthopalli/habitat-api/habitat_baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings/build
[100%] Built target orbslam2
Install the project...
-- Install configuration: "Release"
CMake Error at cmake_install.cmake:50 (file):
  file cannot create directory:
  /opt/anaconda3/envs/HandcraftedAgents/lib/python3.6/site-packages.  Maybe
  need administrative privileges.

Makefile:61: recipe for target 'install' failed
make: *** [install] Error 1
`
Can you please let me know how to resolve this.

mcimpoi, 2019-06-07:
Hi Kowshik,

If you have sufficient space, you can install anaconda in your own home folder, or a folder where you have permissions (e.g. get the Anaconda****.sh script from the website via wget, and run it).

You can try this script to get the dependencies installed:
[https://gist.github.com/mcimpoi/faf9d727b813104fb0c742d002acfa38](https://gist.github.com/mcimpoi/faf9d727b813104fb0c742d002acfa38)

And we are using this conda environment spec:
[https://gist.github.com/mcimpoi/cec8a9c21842155373ed2fea47403d96](https://gist.github.com/mcimpoi/cec8a9c21842155373ed2fea47403d96)

You can create the environment from file (conda env create -f conda_env.yml ... ). The provided conda env supports both habitat sim and habitat api, but you need to follow the install instructions from both project pages (python setup.py / pip install habitat-...); just skip the numpy installation and other requirements (handled by conda).

Hope this helps.

xiaotaw, 2019-11-13:
Hi mcimpoi, 

> You can try this script to get the dependencies installed:
> https://gist.github.com/mcimpoi/faf9d727b813104fb0c742d002acfa38

This script raised an error when installing ORB_SLAM2 within the `habitat` env. The `habitat` env was created under the guidance of habitat-sim and habitat-apt INSTALLATION.
 As the habitat [INSTALL](https://github.com/facebookresearch/habitat-api#installation) suggest python3.6, does it work with python=3.6? 
```
CMake Error at /home/xt/.conda/envs/habitat/share/cmake-3.14/Modules/FindBoost.cmake:2132 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.65.1

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_python-py35

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.

```

mcimpoi, 2019-11-13:
Hi @xiaotaw,

Here is my conda environment under which I tested the script: [https://gist.github.com/mcimpoi/cec8a9c21842155373ed2fea47403d96](https://gist.github.com/mcimpoi/cec8a9c21842155373ed2fea47403d96) 

Also, please note that I am not in sync (i.e. using latest version of the Habitat environment), but it should still work -- the version for which this script was tested is from ~June / July.

To me it looks you need to install lib-boost-dev (or something like this).
This should help:
[https://stackoverflow.com/questions/3808775/cmake-doesnt-find-boost](https://stackoverflow.com/questions/3808775/cmake-doesnt-find-boost)

Also, do you know which of the orbslam dependencies needs boost? 
If not, try to stop the installation after each section in the script:
E.g. put an exit statement here, see if it works (L41); then put exit after the next one...
[https://gist.github.com/mcimpoi/faf9d727b813104fb0c742d002acfa38#file-install_deps-sh-L41](https://gist.github.com/mcimpoi/faf9d727b813104fb0c742d002acfa38#file-install_deps-sh-L41)

xiaotaw, 2019-11-14:
Thank you mcimpoi. I tried in a env created by https://gist.github.com/mcimpoi/cec8a9c21842155373ed2fea47403d96,  however, got the same error at L92.
The error message says it requires `boost_python-py35` to cmake ORB_SLAM2-PythonBindings, but failed to find it.

 `ldconfig -p | grep boost_python`  in my PC:
```
libboost_python3-py36.so.1.65.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libboost_python3-py36.so.1.65.1
libboost_python3-py36.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libboost_python3-py36.so
libboost_python-py27.so.1.65.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
libboost_python-py27.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libboost_python-py27.so
```
Only boost_python-py27 and boost_python-py36 can be found.

Could you please show the result of  `ldconfig -p | grep boost_python` on your machine?

mcimpoi, 2019-11-14:
```
> ldconfig -p | grep boost_python

	libboost_python-py35.so.1.58.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libboost_python-py35.so.1.58.0
	libboost_python-py35.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libboost_python-py35.so
	libboost_python-py27.so.1.58.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.58.0
	libboost_python-py27.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libboost_python-py27.so
```

Can you try and change the CMakeListst.txt file from ```habitat_baselines/slambased/3rdparty/ORB_SLAM2-PythonBindings```, lines 31-33 to match your version?

```
31 │ find_package(PythonLibs 3.5 REQUIRED)
32 │ find_package(NumPy)
33 │ find_package(Boost 1.45.0 REQUIRED COMPONENTS python-py35)
```

(also maybe ask here: the [https://github.com/ducha-aiki/ORB_SLAM2-PythonBindings](https://github.com/ducha-aiki/ORB_SLAM2-PythonBindings), or on the original repository.

xiaotaw, 2019-11-15:
After modifying CMakeListst.txt(L31-33), the installation succeeded. Thank you @mcimpoi !

mathfac, 2019-11-15:
@mcimpoi, thank you for the answers.

mcimpoi, 2019-11-15:
Welcome!

alstar8, 2020-02-13:
@mcimpoi Hi! Do you have a dockerfile with orbslam in it? It would be very nice if you have one and could share it. I tried with habitat docker, but faced a lot of errors while building ORB_SLAM2-PythonBindings

mcimpoi, 2020-05-24:
Hi @alstar8   -- I didn't try it in docker.
Did you try the conda environment mentioned above? 
[https://gist.github.com/mcimpoi/cec8a9c21842155373ed2fea47403d96](https://gist.github.com/mcimpoi/cec8a9c21842155373ed2fea47403d96)

It might be out of date, but might be a good starting point.

zhangyong511, 2022-08-23:
Hello @xiaotaw , I have encountered the same problem as you. How did you solve it

rpartsey, 2022-08-31:
Hi @zhangyong511 
Have you tried modifying CMakeListst.txt(L31-33) as it was suggested [above](https://github.com/facebookresearch/habitat-lab/issues/96#issuecomment-553910936)?

rpartsey, 2023-01-26:
Closing due to inactivity. 
Note, that SLAM-based methods were removed in https://github.com/facebookresearch/habitat-lab/pull/1075.

closed 2023-01-26 by rpartsey