facebookresearch/habitat-lab #103: installing baselines failed

chagri, 2019-06-05:
## Steps to reproduce

  1. post installing habitat-api, try installing baselines using the following command: "python setup.py develop --all"
  2. Env: Ubuntu 16.04, Conda based Python 3.6

## Observed Results

  * What happened?  This could be a description, log output, etc.

## Expected Results
`
Processing dependencies for habitat==0.1.1
Traceback (most recent call last):
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2575, in _dep_map
    return self.__dep_map
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2667, in __getattr__
    raise AttributeError(attr)
AttributeError: _Distribution__dep_map

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/_vendor/packaging/requirements.py", line 90, in __init__
    req = REQUIREMENT.parseString(requirement_string)
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/_vendor/pyparsing.py", line 1617, in parseString
    raise exc
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/_vendor/pyparsing.py", line 1607, in parseString
    loc, tokens = self._parse( instring, 0 )
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/_vendor/pyparsing.py", line 1379, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/_vendor/pyparsing.py", line 3376, in parseImpl
    loc, exprtokens = e._parse( instring, loc, doActions )
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/_vendor/pyparsing.py", line 1383, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/_vendor/pyparsing.py", line 3164, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pkg_resources._vendor.pyparsing.ParseException: Expected stringEnd (at char 5), (line:1, col:6)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2940, in __init__
    super(Requirement, self).__init__(requirement_string)
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/_vendor/packaging/requirements.py", line 94, in __init__
    requirement_string[e.loc:e.loc + 8]))
pkg_resources.extern.packaging.requirements.InvalidRequirement: Invalid requirement, parse error at "'=1.1.0'"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "setup.py", line 96, in <module>
    cmdclass={"install": InstallCommand, "develop": DevelopCommand},
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/setuptools/__init__.py", line 129, in setup
    return distutils.core.setup(**attrs)
  File "/home/chandrak/anaconda3/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/chandrak/anaconda3/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/chandrak/anaconda3/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 66, in run
    super().run()
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/setuptools/command/develop.py", line 36, in run
    self.install_for_development()
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/setuptools/command/develop.py", line 152, in install_for_development
    self.process_distribution(None, self.dist, not self.no_deps)
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/setuptools/command/easy_install.py", line 746, in process_distribution
    [requirement], self.local_index, self.easy_install
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 786, in resolve
    new_requirements = dist.requires(req.extras)[::-1]
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2611, in requires
    dm = self._dep_map
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2577, in _dep_map
    self.__dep_map = self._filter_extras(self._build_dep_map())
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2606, in _build_dep_map
    dm.setdefault(extra, []).extend(parse_requirements(reqs))
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2933, in parse_requirements
    yield Requirement(line)
  File "/home/chandrak/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2942, in __init__
    raise RequirementParseError(str(e))
pkg_resources.RequirementParseError: Invalid requirement, parse error at "'=1.1.0'"
`

## Relevant Code

```
python setup.py develop --all
```

JasonJiazhiZhang, 2019-06-05:
There's an error in `habitat_baselines/rl/requirements.txt`
should be "torch==1.1.0" instead of "torch=1.1.0"

abhiskk, 2019-06-06:
Reopening in case @chagri still faces some issues.

chagri, 2019-06-06:
works for me now.

closed 2019-06-06 by chagri