facebookresearch/habitat-lab #107: Tests fail on python==3.6

dexter1691, 2019-06-06:
## Steps to reproduce

  1. Run tests on python==3.6

## Observed Results

  * What happened?  This could be a description, log output, etc.
Tests failed. I believe this is because of the newly handed PR.
This is the potential cause. 
https://github.com/facebookresearch/habitat-api/blob/master/habitat/core/dataset.py#L206

```
test/test_dataset.py ..FFFFFFF                                                                                                                                                                                                                                                                                         [100%]

==FAILURES ==
__test_filter_episodes __

    def test_filter_episodes():
        dataset = _construct_dataset(100)

        def filter_fn(episode: Episode) -> bool:
            return int(episode.episode_id) % 2 == 0

>       filtered_dataset = dataset.filter_episodes(filter_fn)

test/test_dataset.py:45:
_ _
habitat/core/dataset.py:125: in filter_episodes
    new_dataset = copy.copy(self)
_ _ 
x = <habitat.core.dataset.Dataset object at 0x7ffb2a909ba8>

    def copy(x):
        """Shallow copy operation on arbitrary Python objects.

        See the module's __doc__ string for more info.
        """

        cls = type(x)

        copier = _copy_dispatch.get(cls)
        if copier:
            return copier(x)

        try:
            issc = issubclass(cls, type)
        except TypeError: # cls is not a class
            issc = False
        if issc:
            # treat it as a regular class:
            return _copy_immutable(x)

        copier = getattr(cls, "__copy__", None)
        if copier:
>           return copier(x)
E           TypeError: __copy__() takes 1 positional argument but 2 were given

```

## Expected Results

Tests pass

## Relevant Code

erikwijmans, 2019-06-06:
The CI runs python python==3.6 (https://github.com/facebookresearch/habitat-api/blob/master/.circleci/config.yml#L87) and they are passing there.  Can you try a fresh conda env?

Output from CI on current master:
```
============================= test session starts ==============================
platform linux -- Python 3.6.8, pytest-4.6.2, py-1.8.0, pluggy-0.12.0
rootdir: /home/circleci/project/habitat-api, inifile: setup.cfg, testpaths: test
collected 43 items                                                             

test/test_baseline_agents.py ..                                          [  4%]
test/test_config.py ..                                                   [  9%]
test/test_dataset.py .........                                           [ 30%]
test/test_habitat_env.py ..........s                                     [ 55%]
test/test_habitat_example.py ....                                        [ 65%]
test/test_install.py .                                                   [ 67%]
test/test_mp3d_eqa.py sss                                                [ 74%]
test/test_pointnav_dataset.py .ss.                                       [ 83%]
test/test_sensors.py .....                                               [ 95%]
test/test_trajectory_sim.py ..                                           [100%]

=========================== short test summary info ============================
```

mathfac, 2019-06-07:
@dexter1691 do you still have this issue?

dexter1691, 2019-06-10:
python 3.6.8 fixes the issue. thanks

closed 2019-06-10 by dexter1691