facebookresearch/habitat-lab #108: Clarification for "training steps" in habitat paper and the implementation

takuma-ynd, 2019-06-06:
In Figure 3 of [habitat paper](https://arxiv.org/pdf/1904.01201.pdf), the number of **steps** taken in training and corresponding SPL scores are reported.

I'm basically running `habitat_baselines/train_ppo.py` with the following default params
```
--num-processes 4
--num-steps 128
--num-mini-batch 4
--num-updates 100000
```
Under this setting, how many **steps** does the agent experience?

If I understand the paper and the implementation correctly, the number of **steps** this can experience should be `128 * 100000 * 4 = 51.2M`.

But when I evaluated the trained agent by using `habitat_baselines/evaluate_ppo.py`, the resulting SPL score was way lower than expected. ( I got 0.25 on RGB, whereas Fig3 says it should be above 0.4)

bhavanajain, 2019-06-07:
Can you give us the full commands you are using for training and evaluation. Specifically, the config files.

StOnEGiggity, 2019-06-07:
I also have the same confusion.
I evaluated the agent by ``habitat_baselines/evaluate_ppo.py``, I got similar result(spl 0.258 on RGB). I also evaluated the pretrained model from [habitat-challenge](https://github.com/facebookresearch/habitat-challenge), I got 0.277 on RGB.
This is my train commands:
```
python -u habitat_baselines/train_ppo.py
    --use-gae \
    --sim-gpu-id 1 \
    --pth-gpu-id 1 \
    --lr 2.5e-4 \
    --clip-param 0.1 \
    --value-loss-coef 0.5 \
    --num-processes 4 \
    --num-steps 128 \
    --num-mini-batch 4 \
    --num-updates 100000 \
    --use-linear-lr-decay \
    --use-linear-clip-decay \
    --entropy-coef 0.01 \
    --log-file "logs/train.log" \
    --log-interval 5 \
    --checkpoint-folder "/data/checkpoints" \
    --checkpoint-interval 50 \
    --task-config "configs/tasks/pointnav_gibson.yaml" \
    --sensors "RGB_SENSOR" \
```
And this is my eval commands:
```
python -u habitat_baselines/evaluate_ppo.py \
    --model-path "/data/checkpoint/ckpt.1999.pth" \
    --sim-gpu-id 0 \
    --pth-gpu-id 0 \
    --num-processes 4 \
    --count-test-episodes 100 \
    --task-config "configs/tasks/pointnav_gibson.yaml" \
    --sensors "RGB_SENSOR" \
```

erikwijmans, 2019-06-07:
`--count-test-episodes 100` will only evaluate 100 episodes.  There are 994 episodes in val, looks like you got unlucky with the ones it randomly selected :)

takuma-ynd, 2019-06-07:
@bhavanajain 
I still might want to have an answer for **steps**.
The parameters I'm using for training is basically the default configuration (except for checkpoint interval):
```
python -u baselines/train_ppo.py \
    --use-gae \
    --sim-gpu-id 0 \
    --pth-gpu-id 0 \
    --lr 2.5e-4 \
    --clip-param 0.1 \
    --value-loss-coef 0.5 \
    --num-processes 4 \
    --num-steps 128 \
    --num-mini-batch 4 \
    --num-updates 100000 \
    --use-linear-lr-decay \
    --use-linear-clip-decay \
    --entropy-coef 0.01 \
    --log-file "train.log" \
    --log-interval 5 \
    --checkpoint-folder "data/checkpoints" \
    --checkpoint-interval 100 \
    --task-config "tasks/pointnav_gibson.yaml" \
```
And ths is for evaluation:
```
python -u baselines/evaluate_ppo.py \
    --model-path "/path/to/checkpoint" \
    --sim-gpu-id 0 \
    --pth-gpu-id 0 \
    --num-processes 4 \
    --count-test-episodes 100 \
    --task-config "tasks/pointnav_gibson.yaml" \
```
The configuration file `configs/tasks/pointnav_gibson.yaml` hasn't been touched:
```
ENVIRONMENT:
  MAX_EPISODE_STEPS: 500
SIMULATOR:
  AGENT_0:
    SENSORS: ['RGB_SENSOR']
  HABITAT_SIM_V0:
    GPU_DEVICE_ID: 0
  RGB_SENSOR:
    WIDTH: 256
    HEIGHT: 256
  DEPTH_SENSOR:
    WIDTH: 256
    HEIGHT: 256
TASK:
  TYPE: Nav-v0
  SUCCESS_DISTANCE: 0.2
  SENSORS: ['POINTGOAL_SENSOR']
  POINTGOAL_SENSOR:
    TYPE: PointGoalSensor
    GOAL_FORMAT: POLAR
  MEASUREMENTS: ['SPL']
  SPL:
    TYPE: SPL
    SUCCESS_DISTANCE: 0.2
DATASET:
  TYPE: PointNav-v1
  SPLIT: train
  POINTNAVV1:
    DATA_PATH: data/datasets/pointnav/gibson/v1/{split}/{split}.json.gz
```

So, I want to know
- How many **steps** it is trained
- Which **steps** should I refer to if I want to compare the result with those on habitat paper

takuma-ynd, 2019-06-07:
@erikwijmans 
Is it correct that the evaluation performed to plot Fig.3 in the [paper](https://arxiv.org/pdf/1904.01201.pdf) can be replicated by putting `--count-test-episodes 994`?

erikwijmans, 2019-06-07:
The number of steps of experience a model is trained for can be determined by the number of processes (`--num-processes`), the number of steps per rollout (`--num-steps`), and the number of updates/rollouts (`--num-updates`).  You have the correct math for number of steps of experience, i.e. `128 * 100000 * 4 = 51.2M`, and that is what you should refer to for comparing to results in the paper :)

For recreating val numbers, I you'll need `--count-test-episodes 994` *and* `--num-processes` as we need to fix the splitting logic (#89).  The numbers may not be exactly the same, but they should be close.

mathfac, 2019-06-14:
Do we expect any follow ups on this issue?

takuma-ynd, 2019-06-14:
@mathfac Not for me. I guess you can close it.

closed 2019-06-15 by mathfac

hannahlee66, 2019-07-07:
> `--count-test-episodes 100` will only evaluate 100 episodes. There are 994 episodes in val, looks like you got unlucky with the ones it randomly selected :)

Thanks for your answer, I am wondering how many episodes are there for matterport 3d val ? Thanks in advance!

kowshikthopalli, 2019-07-07:
I think there are 495 val episodes in matterport 3d val. 
You can just load the `<va.json.gz>` file  and see the length of the list returned.

hannahlee66, 2019-07-07:
Many thanks!