facebookresearch/habitat-lab #114: New action space

StOnEGiggity, 2019-06-07:
I notice #98 will break the previous checkpoint(4 actions by default). It's mentioned that a registry for action spaces or permuting the weights could solve this problem. But I may miss the following code. I am confused about this part.

Is there any examples or links about the registry or permuting the weights?

Thanks a lot

mathfac, 2019-06-07:
We need to update our current baselines checkpoints to swap weights on the last layer accordingly to action space change.
 For now you can just swap actions after model prediction step and limit action space to 4/remove action. Sorry for the temporary inconvenience.

erikwijmans, 2019-06-07:
Yes, as part of #98 we reorder stop, move_forward, turn_left, and turn_right from [move_forward=0, turn_left=1, turn_right=2, stop=3] to [stop=0, move_forward=1, turn_left=2, turn_right=3].

We are currently working on a PR to add functionality to _extend_ the action space or change the implementation of current actions: #110 but aren't planning to support changing the mapping of action names to ints for now.

Probably the easiest way to correct any current models is to permute the logits (in our code base, here: https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/rl/ppo/utils.py#L47)
```
import numpy as np

old_to_new_mapping = {0: 1, 1: 2, 2: 3, 3: 0}
old_to_new = np.zeros((4, 4), dtype=np.float32)
for k, v in old_to_new_mapping.items():
    old_to_new[k, v] = 1

print(old_to_new)

batch_size = 8
old_logits = np.random.randn(batch_size, 4)
print(old_logits[0])

new_logits = old_logits @ old_to_new
print(new_logits[0])
```

mathfac, 2020-02-06:
The issue was fixed.

closed 2020-02-06 by mathfac