facebookresearch/habitat-lab #115: Impossible to solve episode

mcimpoi, 2019-06-07:
## Steps to reproduce
Run any of the agents (SLAM / PPO) on episode (X7HyMhZNoso; 25) from MP3D.

## Observed Results
Agent can't reach the goal
[image]

## Expected Results
Existing path from agent starting position to goal position.

erikwijmans, 2019-06-07:
The episode is solve-able but the agent will have to traverse different floors (I believe it has to first go down a set up stairs and then up another set of stairs).  These types of episodes weren't something we intended to generate, but they do happen due to the strangeness of MP3D houses :/

kowshikthopalli, 2019-06-18:
sorry to interject here.. but is there an easy way to know in which floor is the agent's starting position for a given episode ?

mcimpoi, 2019-06-18:
From the episode info: mp3d_val/X7HyMhZNoso_ep 25
Start: -5.279,-0.536,15.294
Goal: 5.994,-0.736,12.765 

It looks to me that the coords are stored as X, Z, Y. @erikwijmans, could you please confirm?

erikwijmans, 2019-06-18:
For datasets like Matterport3D, you can use the semantic annotations to figure out what floor the agent is on. The floor annotations are in the same coordinate frame as the agent in habitat sim (they should be at least). 

@mcimpoi We use -Y as gravity while MP3D uses -Z (we transform all of MP3D on load to be consistent however). I forget off the top of my head how that change (-Z to -Y permitted the axes).

erick84mm, 2019-12-11:
I don't know if this helps but, you can calculate the relative_elevation from pointA to pointB and that will give you the angle of elevation which I believe if it is higher  than pi/6 or lower than -pi/6 you may want to go down or up.

`def get_relative_elevation(self, posA, posB):
        target_vector = np.array(posB) - np.array(posA)
        y = target_vector[1]
        target_vector[1] = 0
        target_length = np.linalg.norm(np.array([target_vector[0], -target_vector[2]]))
        return np.arctan2(y, target_length)`

mathfac, 2020-02-06:
Closing the issue, as it looks resolved.

closed 2020-02-06 by mathfac