facebookresearch/habitat-lab #122: Dockerfile is wrong?

takuma-ynd, 2019-06-13:
## Steps to reproduce

  1. Build an image from the Dockerfile
  2. Create a container from the image 
  3. Run `pip install -r requirements.txt` and `python setup.py develop --all` based on [the description](https://github.com/facebookresearch/habitat-api/tree/master/habitat_baselines).
  4. Run training command written [here](https://github.com/facebookresearch/habitat-api/tree/master/habitat_baselines)

## Observed Results
I got the following error:
```
Traceback (most recent call last):
  File "habitat_baselines/train_ppo.py", line 15, in <module>
    import habitat
  File "/habitat-api/habitat/__init__.py", line 9, in <module>
    from habitat.core.benchmark import Benchmark
  File "/habitat-api/habitat/core/benchmark.py", line 17, in <module>
    from habitat.core.env import Env
  File "/habitat-api/habitat/core/env.py", line 19, in <module>
    from habitat.sims import make_sim
  File "/habitat-api/habitat/sims/__init__.py", line 7, in <module>
    from habitat.sims.registration import make_sim
  File "/habitat-api/habitat/sims/registration.py", line 40, in <module>
    _try_register_habitat_sim()
  File "/habitat-api/habitat/sims/registration.py", line 22, in _try_register_habitat_sim
    from habitat.sims.habitat_simulator import HabitatSim
  File "/habitat-api/habitat/sims/habitat_simulator.py", line 142, in <module>
    class HabitatSim(Simulator):
  File "/habitat-api/habitat/sims/habitat_simulator.py", line 177, in HabitatSim
    ) -> habitat_sim.Configuration:
AttributeError: module 'habitat_sim' has no attribute 'Configuration'
```

## What I suspect
In the Dockerfile, [this line](https://github.com/facebookresearch/habitat-api/blob/05f08725a16e01908b19665d15a6aa621462bb95/Dockerfile#L45) specifies the specific commit of habitat sim. I suspect the specified version (commit) is wrong and not compatible with the habitat-api version.

placeforyiming, 2019-06-20:
Please check this one #134 , also you can delete the git checkout part, which will install the latest version.

rpartsey, 2022-08-31:
Closing the issue because it has not had recent activity (and @placeforyiming provided a valid solution).

closed 2022-08-31 by rpartsey