facebookresearch/habitat-lab #133: Resume from the correct episode using vector_env

takuma-ynd, 2019-06-19:
I am trying to resume training of baseline PPO agent from a saved checkpoint.

It is very easy to load the saved pytorch model.
But how to use `vector_env` to **resume training from the consecutive (or specific) episode** is not clear.

More specifically, we have the following structure in [train_ppo.py](https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/train_ppo.py):
```
...
# reset environments
envs.reset()
...

# main loop
while training_not_finished:
    ...
    envs.step(actions)
    ...
```
If I simply add a model loading code to this, I believe that the loaded model will experience the same exact episodes (i.e., same scenes, same start and goal positions) it has experienced during the previous training in the same order.

Intuitively speaking, I feel like I need to replace `envs.reset()` with something like `envs.reset_and_start_from(<final episode index in the previous training> + 1)`.

abhiskk, 2019-06-19:
We currently don't have the functionality to reproduce the exact order of training data, that being said, you can set the seed value to a random number (something like `542`) and this will make sure you don't see the same data as the `train_ppo` model saw during training.

takuma-ynd, 2019-06-19:
I got it. Thank you!

closed 2019-06-19 by takuma-ynd