facebookresearch/habitat-lab #136: What should I do if I want to explore Replica-Dataset with habitat-api ?

placeforyiming, 2019-06-20:
Already can explore Replica with habitat-sim, so what should I do to use habitat-api on this dataset?

abhiskk, 2019-06-20:
For working with the replica dataset you will have to generate episodes similar to the ones present in `val.json.gz` files for Matterport/Gibson (https://github.com/facebookresearch/habitat-api#task-datasets)

placeforyiming, 2019-06-24:
See:

https://github.com/facebookresearch/habitat-api/issues/137

closed 2019-06-24 by placeforyiming

Danmou, 2019-09-11:
> For working with the replica dataset you will have to generate episodes similar to the ones present in `val.json.gz` files for Matterport/Gibson (https://github.com/facebookresearch/habitat-api#task-datasets)

How would one generate these episodes? Is there a script used to generate episodes for Matterport and Gibson?