facebookresearch/habitat-lab #138: How to implement 'collision' sensor

alexsax, 2019-06-23:
#Hi! The readme mentions being able to detect collisions using a `collision sensor`.

It seems like the information should be available via [this function](https://github.com/facebookresearch/habitat-api/blob/1d392504d9239dbdecdeac27da1d205d9074a147/habitat/sims/habitat_simulator/habitat_simulator.py#L471-L483). How could we go about retrieving this information with a `Sensor`?

erikwijmans, 2019-06-23:
There is currently a measure that counts collisions: https://github.com/facebookresearch/habitat-api/blob/1d392504d9239dbdecdeac27da1d205d9074a147/habitat/tasks/nav/nav_task.py#L380-L381

If you'd like a collision sensor instead (as it's something you intend to be an input to the model or part of a supervised loss), you can implement one with similar logic.  The [`ProximitySensor`](https://github.com/facebookresearch/habitat-api/blob/1d392504d9239dbdecdeac27da1d205d9074a147/habitat/tasks/nav/nav_task.py#L284) is probably a good reference for how to go about structuring a simple sensor.

alexsax, 2019-06-23:
Thanks for the very fast reply! This answers my question :)

closed 2019-06-23 by alexsax