facebookresearch/habitat-lab #142: Create TopDownMap class to incapsulate grid/resolution related logic.

mathfac, 2019-06-26:
We have [TopDownMap measure](https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav_task.py#L403) and [maps utils module](https://github.com/facebookresearch/habitat-api/blob/master/habitat/utils/visualizations/maps.py) to work with topdown maps.

## Observed Results
Measure has inner representation of the top down map (`map array, ._ind_x_min , ...
 _coordinate_max`). That's hard to add additional visuals on top of returned map from measure, as mapping from simulator coordinates system to map is missing.

## Expected Results

That would be cleaner to create TopDownMap class that incapsulates all the coordinates related logic and perform actions on top of TopDownMap and have something like render method to get rgb image in the end of pipeline.

That can make harder passing the class as measure result. In this case we can use attr constructor to unpack and create object instance again.

Not sure how that will affect re-usability.

vkkhare, 2019-07-22:
hi @mathfac I would like to contribute to AI habitat. Is there a discussion platform or can you recommend me some pointers

erikwijmans, 2019-07-23:
Hi @vkkhare -- We have a slack group/channel for development.  Feel free to send me an email (etw @ gatech.edu) and I will send you an invite.