facebookresearch/habitat-lab #143: Add a fog of war to TopDownMap visualization

mathfac, 2019-06-26:
We have [TopDownMap measure](https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav_task.py#L403) and [maps utils module](https://github.com/facebookresearch/habitat-api/blob/master/habitat/utils/visualizations/maps.py) to work with topdown maps.

As our trained agents doesn't have explicit scene map that's hard to understand what part of scene was actually seen by the agent. That would be great to implement a fog of war feature to our map visualization:
[image]

Some options what exactly can be reflected:
- Visited agent's positions with some radius
- Seen part of the scene

dhruvbatra, 2019-06-26:
This would be *awesome* to have!

mathfac, 2019-07-17:
Adressed #160.

closed 2019-07-17 by mathfac