facebookresearch/habitat-lab #145: Quickstart keys don't work.

grahamannett, 2019-06-29:
## Steps to reproduce

  1. Using https://aihabitat.org/habitat-api/tutorials/quickstart.html script
  2. Run with python quickstart_example.py or whatever 
  3. Try using the keys

## Observed Results

Using 
```
FORWARD_KEY="w"
LEFT_KEY="a"
RIGHT_KEY="d"
FINISH="f"
```
doensn't make agent move in right direction.  

## Expected Results

For me (and not using caps, just copied what I had written down), 
```
 A IS FORWARD
 D IS LEFT
 F IS RIGHT
```
The action matches up correctly but the keys seem wrong.  Also using W for forward causes the env.episode_over to become True (so presumably that is what would correspond to F)

## Relevant Code

Easy fix would be to just change the key mappings so w = 1, a = 2, d = 3, f = 0

But I dont understand much of this project yet, it seems like its possibly using 
```
class _DefaultSimulatorActions(Enum):
    STOP = 0
    MOVE_FORWARD = 1
    TURN_LEFT = 2
    TURN_RIGHT = 3
    LOOK_UP = 4
    LOOK_DOWN = 5
```

erikwijmans, 2019-06-29:
Looks like that was hard-coded for some reason and broke when we changed the action mapping.  Fix: https://github.com/facebookresearch/habitat-api/pull/147

We by default only load an agent with 4 actions, but we define a mapping between action name and action number for all 6 actions we have a config for (you can see https://github.com/facebookresearch/habitat-api/blob/master/examples/new_actions.py#L134 for how to change which action space config gets load, the `"V1"` config loads all 6 actions) to improve consistency between different people's models/code.

grahamannett, 2019-06-29:
@erikwijmans hmm okay thanks for that, still trying to figure some stuff out.  

Also your fix is still broken lol

erikwijmans, 2019-06-29:
Whoops!  Forgot to check to make sure the exit key worked :)

mathfac, 2019-07-02:
@grahamannett should fixed now. Right?

grahamannett, 2019-07-23:
@mathfac yeah, thanks

closed 2019-07-23 by grahamannett