facebookresearch/habitat-lab #148: Evaluation and pre-trained models

knzo25, 2019-07-01:
I am currently using commit 43ab7241610804dc971344fa4b7d2b674b6f7045 which is currently the master branch, but I also tried previous commits.

I want to replicate the results from the paper, but running the pre-trained methods provide an average success of 0% with a equally null SPL.

Here is the executed command
```
python -u habitat_baselines/evaluate_ppo.py \
    --model-path "/path/to/checkpoint" \
    --sim-gpu-id 0 \
    --pth-gpu-id 0 \
    --num-processes 1 \
    --count-test-episodes 994 \
    --task-config "configs/tasks/pointnav.yaml"
    --sensors RGB_SENSOR 
```

And an example of the output:
```
2019-07-01 18:54:56,249 Average episode reward: -0.460795
2019-07-01 18:54:56,249 Average episode success: 0.000000
2019-07-01 18:54:56,250 Average episode SPL: 0.000000
```

Additionally, since pulling the master branch, with the evaluation scheme, it sometimes fails when it executes the policy, giving the cryptic:

```
cuDNN error: CUDNN_STATUS_BAD_PARAM
```

Which in this case I think it is because it tries to execute the policy on an empty batch (due to how it does not track ended episodes). To avoid this, for now I just keep tracking ended episodes. I don't think this one last error is to blame for the null performance of the pre-trained models since I had this error before this scheme was implemented.

It would seem to me that along the commits since training these models the policy must have changed a little.

Thank you for the great platform and I appreciate any help !

erikwijmans, 2019-07-01:
For the null performance, this is due to changing which number actions are on.  You can use this script to fix the model weights (we will update the ones probably after refactoring the baselines code):  https://gist.github.com/erikwijmans/398a89d0b793cdb878a903b83ceefad3

> Which in this case I think it is because it tries to execute the policy on an empty batch (due to how it does not track ended episodes).

Yeah, we need to add a `or envs.num_envs > 0` to the while loop to keep that from happening :)

JasonJiazhiZhang, 2019-07-03:
> cuDNN error: CUDNN_STATUS_BAD_PARAM

For this error, it happens because CNN module took in a batch of size 0. This is because evaluate_ppo.py checks if episodes are seen and will return batch of size 0 if there's no  unseen episodes (based on episode_id).

After dataset creation, can you do a `print([ep.episode_id for ep in dataset.episodes])` to see if there are 994 unique episode_ids?

mathfac, 2019-07-04:
@knzo25, thank you for creating the issue. You can use `v0.1.1` for habitat-api and habitat-sim repo to replicate tech report results. That should be compatible with pre-trained baselines.  I created relevant issue to add CI test for this failure in the future: https://github.com/facebookresearch/habitat-api/issues/152.

knzo25, 2019-07-06:
Thank you for your answers!
Sorry I did not reply before, but I was retraining the baseline training just in case.

@erikwijmans your script worked perfectly, thank you!
Depth-only agent:

    2019-07-07 07:47:11,079 Average episode reward: 13.620850 
    2019-07-07 07:47:11,080 Average episode success: 0.701873 
    2019-07-07 07:47:11,080 Average episode SPL: 0.584454

However, when I trained the depth-only agent with:
python -u habitat_baselines/train_ppo.py     --use-gae     --sim-gpu-id 0     --pth-gpu-id 1     --lr 2.5e-4     --clip-param 0.1     --value-loss-coef 0.5     --num-processes 6     --num-steps 128     --num-mini-batch 4     --num-updates 100000     --use-linear-lr-decay     --use-linear-clip-decay     --entropy-coef 0.01     --log-file "train.log"     --log-interval 5     --checkpoint-folder "data/checkpoints"     --checkpoint-interval 50     --task-config "configs/tasks/pointnav_depth_mp3d.yaml" --tensorboard-dir logdir

which I believe reproduces the paper results according to #108 (128 * 100000 * 6 gets about the 75 million steps mentioned in the paper), gets:

    2019-07-07 01:10:01,403 Average episode reward: 8.094588
    2019-07-07 01:10:01,403 Average episode success: 0.511579
    2019-07-07 01:10:01,403 Average episode SPL: 0.366799

It turns out that even if I use only DEPTH_SENSOR in pointnav_depth_mp3d.yaml, RGB_SENSOR is still being added (which makes the previous result more consistent with the reported results, as it is in fact the RGB agent), 

the config file:

    ENVIRONMENT:
      MAX_EPISODE_STEPS: 500
    SIMULATOR:
      AGENT_0:
        SENSORS: ['DEPTH_SENSOR']
      HABITAT_SIM_V0:
        GPU_DEVICE_ID: 0
      RGB_SENSOR:
        WIDTH: 256
        HEIGHT: 256
      DEPTH_SENSOR:
        WIDTH: 256
        HEIGHT: 256
    TASK:
      TYPE: Nav-v0
      SUCCESS_DISTANCE: 0.2
      SENSORS: ['POINTGOAL_SENSOR']
      POINTGOAL_SENSOR:
        TYPE: PointGoalSensor
        GOAL_FORMAT: POLAR
      MEASUREMENTS: ['SPL']
      SPL:
        TYPE: SPL
        SUCCESS_DISTANCE: 0.2
    DATASET:
      TYPE: PointNav-v1
      SPLIT: train
      DATA_PATH: data/datasets/pointnav/mp3d/v1/{split}/{split}.json.gz

If I use --sensor DEPTH_SENSOR when running the script it seems to be loading just the depth sensor. Is this the expected behavior?

StOnEGiggity, 2019-11-29:
Hi, is there any reason for worse performance except the order of actions. I am evaluating [SplitNet](https://github.com/facebookresearch/splitnet), which is based on habitat-api.git@7015813aefae99233864c4ffcf7e52e9097392a4. 

After I change to the latest version of habitat, the pre-train models perform worse. However, SplitNet only uses three actions(forward, left and right),
which is unrelated to the above problem. 

This problem confused me for a long time. I'm still unfamiliar with many details about habitat-api and habitat-sim. I look forward to your valuable advice. Thanks a lot.

erikwijmans, 2019-11-29:
One major change that has happened is the renaming with PointGoal Sensors that happened in PR #180  Use `POINTGOAL_WITH_GPS_COMPASS_SENSOR` like here: https://github.com/facebookresearch/habitat-api/blob/master/configs/tasks/pointnav_gibson.yaml#L18

StOnEGiggity, 2019-11-30:
Thank you!
I change my config with `POINTGOAL_WITH_GPS_COMPASS_SENSOR `. The results become normal.

@erikwijmans  Thank you again!

rpartsey, 2022-08-31:
@knzo25 `habitat-lab` has been updated a lot since your las comment.
Closing the issue as stale.

closed 2022-08-31 by rpartsey