facebookresearch/habitat-lab #149: Multiprocess evaluate_ppo.py broken

JasonJiazhiZhang, 2019-07-02:
Multiple process evaluation for PPO was introduced in PR #127, but is currently broken due to buggy update of `test_recurrent_hidden_state`. Current code for calculating `episode_rewards` and `episode_counts` is also buggy for multiprocess. 
For now please use single-process eval for ppo (with flag `--num-processes 1`)

Working on a fix

closed 2019-07-07 by JasonJiazhiZhang

chenwydj, 2019-09-29:
is this issue solved? when I use `--num-processes 2` the multiprocess still raise BrokenPipeError

erikwijmans, 2019-09-29:
This was a bug in logic and didn't cause `BrokenPipeError`.  Please open a new issue for this and provide the additional details it asks for.

chenwydj, 2019-09-29:
> This was a bug in logic and didn't cause `BrokenPipeError`. Please open a new issue for this and provide the additional details it asks for.

Did you mean this is a logic bug in my side? What did you mean by "didn't cause `BrokenPipeError`"?

Thank you!

erikwijmans, 2019-09-30:
May be a logic error on your side, may be a new/different bug.  The bug reported here didn't cause a BrokenPipeError so I doubt this has anything to do with that.