facebookresearch/habitat-lab #157: More agent actions

weihaosky, 2019-07-15:
In habitat-sim we can use W/A/S/D keys to move forward/left/backward/right and arrow keys to control gaze direction (look up/down/left/right).  But in the habitat-api example.py it seems there is only `['move_forward', 'turn_left', 'turn_right']` in the `action_space.key()`. How should I command the agent to move left or right and do other actions?
Many thanks!

weihaosky, 2019-07-15:
solved

erikwijmans, 2019-07-17:
You can add variants of the movement actions such that they only control the camera (as controls which only control the cameras currently have no collisions)  i.e. you re-register the `MoveRight` control with `body_action=False` and it'll only control the camera:

```
import habitat_sim
from habitat_sim.agent import default_controls

habitat_sim.register_move_fn(default_controls.MoveRight, name="camera_move_right", body_action=False)
```

then you can follow the following example to use those actions in the API: https://github.com/facebookresearch/habitat-api/blob/master/examples/new_actions.py

weihaosky, 2019-07-17:
Thanks @erikwijmans ! But when the `body_action` is set to False, it seems the `MoveRight` is no longer `MoveRight`. It is moving to some strange direction. The original `MoveRight` with `body_action=True` is just moving along the x axis. This may be caused by the axis difference between the camera and body.

weihaosky, 2019-07-17:
Oh, it seems this also happens for `body_action=True`. After the agent turns, the `MoveRight` still follows the original axis direction, which mean it is no longer a "moveright". How to let the agent move along the current right direction?

erikwijmans, 2019-07-17:
Can you try using this branch on habitat-sim: https://github.com/facebookresearch/habitat-sim/tree/fix-move

That should fix the issue with `move_right` not moving in the correct coordinate frame.

weihaosky, 2019-07-18:
Yes, change `absolute_transformation` to `transformation` works!

weihaosky, 2019-07-18:
But when the camera moves away from the agent body, the turn_left or turn_right still rotates around the agent body rather than the camera.

erikwijmans, 2019-07-18:
You’ll need to use `look_left` instead.

weihaosky, 2019-07-19:
`look_left` will make the two cameras in a stereo structure rotate respectively. Thus after `look_left` it will no longer be stereo structure. So it `look_left` cannot be used in stereo camera.

erikwijmans, 2019-07-19:
Given that you want to control a stereo pair, the body actions make more sense as that'll control a point in the middle of them.  You can disable collision checking by either setting `apply_filter` to false here: https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/agent/agent.py#L158 or commenting this loop: https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/simulator.py#L108-L109

weihaosky, 2019-07-19:
Yeah! That's what I am looking for. Thanks man!

mathfac, 2020-01-10:
Closing as resolved.

closed 2020-01-10 by mathfac