facebookresearch/habitat-lab #158: How to get disparity map

weihaosky, 2019-07-15:
Hi, how should I get the disparity map for the stereo agent? I can get depth but do not know how to convert it to disparity map. The baseline is known but focal length unknown. How to get the focal length in pixel?

Many thanks!

xmyqsh, 2019-07-15:
@weihaosky From the intrinsic matrix

weihaosky, 2019-07-15:
@xmyqsh Thanks for reply. How can I get the intrinsic matrix?

xmyqsh, 2019-07-15:
@weihaosky
Use the classic chessboard method.
Or, if this way is not available for you, 
RGB-D camera? You can estimate it by PNP with BA. Just add the intrinsic matrix as an optimized parameter.

weihaosky, 2019-07-15:
@xmyqsh 
I mean in habitat-api how to get the disparity map. For now the output of habitat-sim is only depth map rather than disparity map. I want to get the ground truth disparity map for two cameras, e.g. the stereo agent in the example

xmyqsh, 2019-07-15:
@weihaosky 
...
You can calculate it with the default `FOV = 35 degree` in the [PinholeCamera](https://github.com/facebookresearch/habitat-sim/blob/master/src/esp/sensor/PinholeCamera.h#L44)

And the [formulation](https://doc.magnum.graphics/magnum/classMagnum_1_1Math_1_1Matrix4.html#a6475b3ef155c9142b890c8133504ae9b) from `FOV` to `focal length` (the aspect ratio should be 1 in the stereo agent example) or `focal length / depth`.

erikwijmans, 2019-07-15:
Yep, the intrinsic camera matrix is just a simple pinhole model.  The default FOV is actually 90 degrees and can be configured to anything you want by setting the corresponding value in the config: https://github.com/facebookresearch/habitat-api/blob/master/habitat/config/default.py#L100

mathfac, 2019-07-15:
Thank you @xmyqsh and @erikwijmans for the right answers and growing community! There is an example of intrinsic parameters calculations in [our notebook example](https://github.com/facebookresearch/habitat-api/blob/master/notebooks/relative_camera_views_transform_and_warping_demo.ipynb) cell # 5:
```
K = np.array([
    [1 / np.tan(hfov / 2.), 0., 0., 0.],
    [0., 1 / np.tan(hfov / 2.), 0., 0.],
    [0., 0.,  1, 0],
    [0., 0., 0, 1]])
```

weihaosky, 2019-07-16:
According to @xmyqsh , the intrinsic matrix should be 
```
K = np.array([
    [1 / tan(hfov / 2.), 0., 0., 0.],
    [0., a / tan(hfov / 2.), 0., 0.],
    [0., 0.,  (n+f)/(n-f), 2nf/(n-f)],
    [0., 0., -1, 0]])
```
So the focal length fx in pixel should be `fx = 1 / tan(hfov / 2) * Width / 2`
Then the disparity should be `d = Baseline * fx / depth`
Am I right?

erikwijmans, 2019-07-16:
Yes, the matrix @mathfac linked is in OpenGL's clip-space coordinate system while the one from @xmyqsh is in pixel space.

weihaosky, 2019-07-16:
Thank everyone!

closed 2019-07-22 by weihaosky