facebookresearch/habitat-lab #165: Is it possible to utilise a continuous action space?

krishanrana, 2019-07-18:
I would like to experiment an idea with TD3 and wanted to know if it would be possible to use a continuous action space for the agent, and how to got about setting it up.

abhiskk, 2019-07-25:
hi, can you give more details about what exactly you mean by a continuous action space?

krishanrana, 2019-07-25:
Hi abhishek, I would like to utilise an action space consisting of linear velocity and angular velocity as opposed to the discrete actions (turn left, go forward etc) in order to control my agent in the environment.

erikwijmans, 2019-07-25:
The core of habitat-sim does support continuous action spaces (the collision detection logic makes no assumptions about discrete actions) but the API to properly support continuous actions (both habtiat-api and in the simulator itself) aren't there.

huangxueqin123, 2019-08-07:
Hi krishanrana, I also have the same problem.Have you solved it?

> Hi abhishek, I would like to utilise an action space consisting of linear velocity and angular velocity as opposed to the discrete actions (turn left, go forward etc) in order to control my agent in the environment.

mathfac, 2019-10-22:
Hi @huangxueqin123 and @krishanrana,
We have continuous actions space for [habitat-api Task Actions](https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav.py#L842). You can add your own task action `move(velocity, angular_velocity)` and set/calculate agent position accordingly in `Task.step` method.

Our plan is to add native support of continues actions to the Habitat Simulator till the end of the year.

cc @aclegg3

Skylion007, 2020-08-04:
@mathfac Do we have any updates on this front?

aclegg3, 2020-08-04:
We have an example of using continuous actions in Habitat-sim in [this tutorial](https://aihabitat.org/docs/habitat-sim/rigid-object-tutorial.html#continuous-control-on-navmesh). You'd just need to implement the task action wrapper for this I think.

rpartsey, 2022-09-01:
@ASzot do I understand correctly that such functionality is already implemented in habitat-lab as the [BaseVelAction](https://github.com/facebookresearch/habitat-lab/blob/380ac0a7d8c4ead1532f109b15d329473212eae9/habitat/tasks/rearrange/actions/actions.py#L227)?

ASzot, 2022-09-01:
@rpartsey that's correct. Continuous actions are very well integrated in Habitat now.

closed 2022-09-04 by rpartsey