facebookresearch/habitat-lab #169: Identify the current region of the agent/ pointers for room navigation

kowshikthopalli, 2019-07-21:
Hi,
Is there an easy way to identify the current region/room the agent is in.? . through the example notebook 
"""
env.sim.get_agent_state() 
"""
would only give the agent's position and rotation. 
This information is needed to implement a `room sensor` for a `room navigation` task.  

TIA.

erikwijmans, 2019-07-22:
`sim. semantic_annotations.regions()` returns a list of regions.  Each region has an `aabb` field, which returns an axis aligned bounding box.  You can check to see if the agent is within that bounding box or not.

kowshikthopalli, 2019-07-22:
Thank you very much Erik for the reply.  I am aware of the method `env.sim.semantic_annotations()` as it is also used in the demo notebook. 
I wasn't sure if I could directly call that function in a given episode while the agent is taking actions. 
Also is there any small snippet of code that you could share to do the check with `aabb`.

erikwijmans, 2019-07-22:
Yes, you can call that function during episodes.

I haven't tested this, but this should do the check:
```python
np.all(
    ((aabb.center - aabb.sizes.abs() / 2) <= pt) & 
    ((aabb.center + aabb.sizes.abs() / 2) >= pt))
```

josselinbo, 2019-07-24:
Hi,  

Is there a way to directly get the level at which the agent is located ? Or do also we have to compare position with the bounding boxes of the levels ? 

Thank you in advance

kowshikthopalli, 2019-07-24:
I am not sure if you can directly get the level in which the agent is in.
You are right about the second part. Checking with each level's aabb center
and sizes will give you the correct answer.

josselinbo, 2019-07-24:
Thank you for your answer.

erikwijmans, 2019-07-24:
No, we don't currently have a method that retrieves that for you (such a method would just check all the bounding boxes internally tho).

closed 2019-07-25 by kowshikthopalli

qmaruf, 2020-09-07:
Is there any way to get the field of view region name? Or the filed of view coordinates?

Mayer123, 2021-03-31:
Hi, 

When I tried to print out the aabb of the regions, the aabb.center looks reasonable, but the aabb.sizes are all 0 (e.g. all regions in D7N2EKCX4Sj). I'm wondering if this is expected? This is how I check it
scene= self.env.sim.semantic_annotations()
regions = scene.regions
for region in regions:
        print (region.id, region.category.name(), region.aabb.center, region.aabb.sizes)
Thanks!

zillur01, 2026-01-02:
HI @Mayer123 were you able to solve that? for my case, `region.aabb.sizes` shows all `inf` while `regions.aabb.center` shows all 0