facebookresearch/habitat-lab #174: Is there a mapping between Matterport Viewpoints ids and episodes ids?

erick84mm, 2019-07-26:
## Steps to reproduce
I am trying to add the Room to Room (R2R) data set to the habitat-API since I want to work with the instructions provided in that data set. In the R2R the data is provided with the following format:

`{
    'distance': 11.66, 
    'scan': 'VLzqgDo317F', 
    'path_id': 6250, 
    'path': [
        'af3af33b0120469c9a00daa0d0b36799', 
        '5be145994f974347850a48cecd04cdcd', 
        '79aedad1206b4eea9c4b639ea2182eb7', 
        '1c91ed40af2246f2b126dd0f661970df', 
        '385019f5d018430fa233d483b253076c', 
        'fd263d778b534f798d0e1ae48886e5f3'
    ], 
    'heading': 3.751, 
    'instructions': [
        'Walk down one flight of stairs and stop on the landing. ', 
        'Walk between the columns and make a sharp turn right. Walk down the steps and stop on the landing. ', 
        'walk forward then turn right at the stairs then go down the stairs. '
    ]
}`

So I was wondering if there is a mapping of path ids ('path_id': 6250, ) to episode id or perhaps the viewpoint ids ('af3af33b0120469c9a00daa0d0b36799').

If there is no mapping, how can I translate the camera intrinsics, heading and elevation to the quartenions required for the pointNav task.

PD: Thank you very much for your help in advance.

erikwijmans, 2019-07-26:
There isn't a mapping for R2R, although developing one would be great and we are happy to assist in this!

Mapping the viewpoints to habitat-sim coordinates would be the first place to start.  We apply a transformation to MP3D to make +Y up instead of +Z.  That transform is specified by the quat `habitat_sim.utils.quat_from_two_vector(np.array([0, 0, 1]), np.array([0, 1, 0])`

erick84mm, 2019-07-26:
Ok I understand, so we may need to take the heading and camera intrinsics for each viewpoint and do the transformation that you describe for each of them. Thank you for the clarification I will start by doing that.

erikwijmans, 2019-07-27:
I don't believe you'll need the camera intrinsics for anything

rpartsey, 2022-09-01:
Closing the issue because it has not had recent activity.

closed 2022-09-01 by rpartsey