facebookresearch/habitat-lab #177: How can I evaluate habitat-challenge code under habitat-api environment without the docker?

hannahlee66, 2019-08-01:
Seems this habitat-challenge code (https://github.com/facebookresearch/habitat-challenge) needs to build docker, how can we run it under the official habitat-api / habitat-sim environment without the docker?

Thanks a lot!

abhiskk, 2019-08-02:
You can use the code from [agent.py](https://github.com/facebookresearch/habitat-challenge/blob/master/agent.py) to evaluate this.

PS: you might have to make some changes since the habitat-challenge code has not been updated with the latest releases, happy to help out on this here.

hannahlee66, 2019-08-02:
> You can use the code from [agent.py](https://github.com/facebookresearch/habitat-challenge/blob/master/agent.py) to evaluate this.
> 
> PS: you might have to make some changes since the habitat-challenge code has not been updated with the latest releases, happy to help out on this here.

Many thanks!

Yes. As you mentioned, the latest release is not the same, so when I apply it to habitat-api, SPL = 0, while in this code, I can successfully get 0.9. Is there any solution to solve this?

erikwijmans, 2019-08-02:
Two things have changed.  We moved stop to action 0 and the baselines code got a refactor.  You can use this script to fix the weights in the softmax layer and change the names of weights to fix the new structure: https://pastebin.com/dVNKdHSA

hannahlee66, 2019-08-03:
> Two things have changed. We moved stop to action 0 and the baselines code got a refactor. You can use this script to fix the weights in the softmax layer and change the names of weights to fit the new structure: https://pastebin.com/dVNKdHSA

Oh! Thanks, will try first to see.

closed 2022-09-01 by rpartsey