facebookresearch/habitat-lab #181: Disabling instance meshes for Matterport 3D

srama2512, 2019-08-05:
## Observed Results
While running experiments on Matterport3D, the instance meshes are also being loaded without me explicitly asking for the semantic sensor. It's significantly increasing the memory consumption for no reason. Is there a way to disable this?

srama2512, 2019-08-05:
My current solution is to create a copy of scene_datasets/mp3d without *_semantic.ply files. The check seems to be happening within Simulator.cpp and I don't see a straightforward way to disable that conditionally.

abhiskk, 2019-08-29:
Apologies for the delayed reply, not adding the semantic sensor [here](https://github.com/facebookresearch/habitat-api/blob/master/configs/tasks/pointnav.yaml#L5) should work.

closed 2020-01-03 by srama2512