facebookresearch/habitat-lab #184: VectorEnv destruction error when tensorboard enabled

JasonJiazhiZhang, 2019-08-08:
## Steps to reproduce

```bash
python -u habitat_baselines/run.py --exp-config habitat_baselines/config/test/ppo_pointnav_test.yaml --run-type train 
```
runs fine but enabling tensorboard logging (adding `TENSORBOARD_DIR "tb"` at the end) would result in this error:
```
Exception ignored in: <function VectorEnv.__del__ at 0x7fb8194462f0>
Traceback (most recent call last):
  File "/private/home/jiazhi/habitat-api/habitat/core/vector_env.py", line 458, in __del__
  File "/private/home/jiazhi/habitat-api/habitat/core/vector_env.py", line 336, in close
  File "/private/home/jiazhi/.conda/envs/fair_jiazhi/lib/python3.7/multiprocessing/connection.py", line 206, in send
AttributeError: 'NoneType' object has no attribute 'dumps'
```

This is likely due to tensorboard file writer prematurely closing IO buffers that was still used by VectorEnv.

erikwijmans, 2019-08-08:
If you explicitly close the vector envs before exiting the TB writer context, does that fix it?

JasonJiazhiZhang, 2019-08-08:
> If you explicitly close the vector envs before exiting the TB writer context, does that fix it?

That seems to fix it :tada: thanks for the suggestion!

closed 2019-08-29 by abhiskk