facebookresearch/habitat-lab #186: Can Habitat run on OpenGL 3.3?

UncleMEDM, 2019-08-12:
After I install habitat-sim and habitat-api, and run the command python ./examples/exmaple.py, I encounted the following error:

(mp3d) haolin@local:/data1/haolin/habitat-api$ python examples/example.py
2019-08-12 18:21:17,177 initializing sim Sim-v0
Renderer: Gallium 0.4 on llvmpipe (LLVM 3.8, 256 bits) by VMware, Inc.
OpenGL version: 3.3 (Core Profile) Mesa 17.0.0
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    mesa-forward-compatible-line-width-range
Magnum: required version OpenGL 4.1 is not supported
Aborted (core dumped)

It seems like that only OpenGL >=4.1 can run the program? But I don't want to mess with the graphics card driver since this server is used by a lot of my teammates. Is there any solution for this? Thank you!

xiaotaw, 2019-11-13:
You can use habitat in a docker container with OpenGL >= 4.1.

rpartsey, 2022-09-01:
Closing the issue because it has not had recent activity.

closed 2022-09-01 by rpartsey