facebookresearch/habitat-lab #189: Shortest path follower is inconsistent with SPL

jozhang97, 2019-08-16:
Hi, 

The path follower's and SPL's exit conditions differ: one uses euclidean distance, the other geodesic. 

Here is shortest path follower using euclidean distance: https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/shortest_path_follower.py#L71

Here is SPL using geodesic distance: https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav_task.py#L426

I think if someone is trying to improve SPL while using shortest path follower's it would be better to use the exact same metric (geodesic distance in this case).

abhiskk, 2019-08-29:
Hi @jozhang97, the usage of cartesian distance in path follower (different from geodesic distance in SPL calculation) is correct. This is because for calculating the geodesic distance you need access to the underlying visitation map of environment and technically this information is not available for the agent moving around in the environment. The only information available to agent is the pointgoal coordinates to target location and sensor inputs [RGB, Depth].

erikwijmans, 2019-08-29:
The shortest path follower has access to the map -- it is an oracle :)

mathfac, 2019-08-29:
It looks like valid concern/issue as stop/success criteria for Shortest Path Follower is different from SPL metrics.
As a right way to fix, we can change Shortest Path Follower stop criteria to use geodesic distance:
```
        if (self._sim.geodesic_distance(self._sim.get_agent_state().position, goal_pos)
            <= self._goal_radius
        ):
            return self._get_return_value(SimulatorActions.STOP)
```
That may affect speed of the Path Follower.

jozhang97, 2019-08-29:
Thanks for the responses :)

I had made the change @mathfac suggested in my code some time ago. It helped in some edge cases with different floors.

erikwijmans, 2019-08-29:
The issue with different floors shouldn't matter unless the goal radius is very large.

@mathfac -- The follower already runs the shortest path algo, so we can re-uses that, but it really doesn't matter as the shortest path algo takes like 10 microseconds in habitat.

mathfac, 2019-08-29:
@erikwijmans, I am sending the proposed fix to keep consistency and cover any edge cases with walls/floors/overloaded spaces.

closed 2019-08-29 by mathfac