facebookresearch/habitat-lab #190: pointgoal measurement doesn't change in macOS

Toraudonn, 2019-08-21:
## Steps to reproduce / Observed Results

I was playing around with the [quickstart script](https://aihabitat.org/habitat-api/tutorials/quickstart.html) and found that the `observations["pointgoal"]` doesn't change when I use habitat-api from my macOS (Mojave). It works flawlessly on my ubuntu desktop so I was wondering if you know why this occurs. My intuition was that habitat-sim is cross-platform so habitat-api should work also. Thanks in advance!

My observations:
- `observations["pointgoal"]` (both distance and theta) doesn't change each step
- `observations["rgb"]` does change each step (it renders fine)

erikwijmans, 2019-08-21:
I forgot to update the example script!  We changed the naming semantics of the goal sensors to fit the semantics of the task in #180.  The old functionality of the `pointgoal` sensor is now the `pointgoal_with_gps_compass` sensor.  The config in the example script should have been updated however, so I am surprised you aren't getting a dictionary key error.

erikwijmans, 2019-08-21:
@abhiskk -- How do I update the example script?

Toraudonn, 2019-08-21:
@erikwijmans `pointgoal_with_gps_compass` works! I didn't catch that in the config files.

https://github.com/facebookresearch/habitat-api/blob/b5f2b00a25627ecb52b43b13ea96b05998d9a121/habitat/config/default.py#L39

Is this somehow related? I was glancing though this while trying to figure it out.

closed 2019-10-11 by Toraudonn