facebookresearch/habitat-lab #202: Where to get data split for eqa task

lynl7130, 2019-09-15:
Hi, 

I want to use Habitat for ObjectGoal Navigation. 

Where could I get the data split of eqa (looks like object goals are provided) for Matterport3D? 

I see [this file](https://github.com/facebookresearch/habitat-api/blob/d69bf69cbc7afbd286583c2d5b8c2afb97cb1eb9/habitat/datasets/eqa/mp3d_eqa_dataset.py) uses the split from directory:

```
.../habitat-api/data/datasets/eqa
```

Thanks in advance!

mathfac, 2019-09-20:
Hi @lynl7130, we are working on releasing those data files and plan to release it before 10-th of October. If you need any specific information right now as scene splits or object names list I can provide it.

lynl7130, 2019-09-21:
Thank you @mathfac ! October 10 is nice. I would wait for your release.

closed 2019-09-21 by lynl7130