facebookresearch/habitat-lab #205: [Feature Request] High resolution videos

erikwijmans, 2019-09-18:
## Feature Request

With https://github.com/facebookresearch/habitat-sim/pull/114 landed, habitat-sim supports sensors of multiple resolutions.  This means the habitat-api video generation code can be updated to output videos at arbitrary (and higher!) resolutions.

## Steps

1. Add sensor uuid to the config -- Each sensor needs to have a different name, so if you need a lowres and a highres RGB sensor, they will need to have different names and this should be specified in the config.
1. Add resolution as a parameter in the habitat_baselines config
    1. Use that to set the resolution of the top-down map
    1. Use that to set the resolution of the video RGB sensor.
1. Update habitat-api video generation code to be fine with any resolution.

erikwijmans, 2019-10-24:
You are certainly welcome to work on this!

I have hacked this together very roughly once before.  The place to is adding a higher-res RGB sensor to the base config, like:
```python
# -----------------------------------------------------------------------------
# # RGB SENSOR
# -----------------------------------------------------------------------------
_C.SIMULATOR.RGB_SENSOR = SENSOR.clone()
_C.SIMULATOR.RGB_SENSOR.TYPE = "HabitatSimRGBSensor"
_C.SIMULATOR.RGB_SENSOR.UUID = "rgb"

_C.SIMULATOR.VIDEO_RGB_SENSOR = _C.SIMULATOR.RGB_SENSOR.clone()
_C.SIMULATOR.VIDEO_RGB_SENSOR.UUID = "video_rgb"
_C.SIMULATOR.VIDEO_RGB_SENSOR.WIDTH = 1024
_C.SIMULATOR.VIDEO_RGB_SENSOR.HEIGHT = 1024
```

You can then add a `VIDEO_RGB_SENSOR` to the list of sensors and pull its observation for the video.

The base sensors then need to be updated to get their uuid (their name) from the config: https://github.com/facebookresearch/habitat-api/blob/master/habitat/core/simulator.py#L112

I'd start with updating this example: https://github.com/facebookresearch/habitat-api/blob/master/examples/shortest_path_follower_example.py

To do high-res video, and then we can move that into habitat-baselines also!

Yash621, 2021-11-05:
@erikwijmans  can I work on this ?

erikwijmans, 2021-12-06:
Yep! Feel free to.