facebookresearch/habitat-lab #206: Question about PointNav data info

DianDYu, 2019-09-19:
In the provided PointNav datasets for Gibson and MatterPort3D, how do you define the episode difficulty for easy, medium, and hard? By geodesic distance or by the ratio between geodesic and euclidean distance? I didn't find this information in the code and the paper. Thanks!

```
// TODO(you): code here to reproduce the problem
```

kowshikthopalli, 2019-09-19:
Hi,
Episode difficulty is defined as the ratio of geodesic shortest path position to Euclidean distance. More detailed comments can be found here 
https://github.com/facebookresearch/habitat-api/blob/959799b948dd7d11d72ea417d47f0c5f60e35fc2/habitat/datasets/pointnav/pointnav_generator.py#L94

DianDYu, 2019-09-19:
Here, similar to the paper, shows how to do rejection sampling. However, what's the ratio/boundary used to decide what games are easy, medium, and hard, respectively? Thanks!

erikwijmans, 2019-09-19:
The easy, medium, and hard ratings are something we had originally considered using, but creating one that were consistent across a variety of scene datasets (consistent across Gibson, Matterport3D, Replica, etc) was impossible and thus they would just be arbitrary for each scene dataset and not useful.  The difficult ratings that are in the episodes shouldn't be used for anything.

Our preferred method for doing difficulty analysis is to plot SPL vs. geodesic_distance as this a) gives more fine-grained analysis and is b) a consistent thing to do across a variety of scene datasets.

DianDYu, 2019-09-20:
@erikwijmans Thanks for the explanation! To be consistent with the training/val dataset where you defined easy/medium/hard, can I generate the same difficulty level for the episodes in the test dataset?

erikwijmans, 2019-09-20:
The labeling of easy/medium/hard was something we were attempting to do with episodes after generation.  Those labels didn't influence the generation in anyway -- the script in habitat-api is the one used to generate the episodes in the released train/val splits.

DianDYu, 2019-09-20:
@erikwijmans Thanks! I'm trying to evaluate my models on the test set to see how they perform on different level of difficulties (as they show some patterns on the val set). It would be great if you can point out how to assign "easy", "medium", and "hard" labels to each game in the test set, the same way as that for train/val set on Matterport3D. Thanks!

erikwijmans, 2019-09-20:
We don't have a way to assign those labels as we never converged on anything so I (nor anyone) has an idea of exactly what way those labels were assigned -- I don't believe we intended them to be released.

For performance vs. difficultly, I would recommend plotting SPL vs. geodesic distance.

mathfac, 2019-09-20:
@DianDYu, as @erikwijmans mentioned difficulties are redundant information that wasn't intended to use for any type of analysis. But to answer your particular question it was assigned based on geodesic distance range with next intervals:
```
DIFFICULTIES_BASED_GEODESIC_DISTANCE = {
    0: {
        "difficulty": "easy",
        "start": 1,
        "end": 7,
    },
    1: {
        "difficulty": "medium",
        "start": 7,
        "end": 13,
    },
    2: {
        "difficulty": "hard",
        "start": 13,
        "end": 24,
    },
}
```
Hope we answered your question in full.

DianDYu, 2019-09-20:
@mathfac @erikwijmans Thank you very much for your answers and suggestions!

closed 2019-09-20 by DianDYu