facebookresearch/habitat-lab #217: Simulator reconfigure require different scene.

dexter1691, 2019-09-28:
## ❓ Questions and Help

https://github.com/facebookresearch/habitat-api/blob/master/habitat/sims/habitat_simulator/habitat_simulator.py#L271

Why do we require the scene_id to be different when trying to reconfigure the Simulator? The current implementation doesn't cover cases where I might want to reconfigure the simulator on the fly with different actuation spec for the same scene.

erikwijmans, 2019-10-03:
Looks like we didn't consider this case.  When swapping the scene, it makes sense to just kill the whole simulator and bring up a new one for that scene.

Does adding a
```python
else:
    self._sim.reconfigure(self.sim_config)
```
do it?

dexter1691, 2019-10-13:
Yeah. That should fix it.

closed 2019-10-13 by dexter1691