facebookresearch/habitat-lab #223: How to get the agent's current position (x, y) and camera viewpoint?

chenwydj, 2019-10-03:
## ❓ Questions and Help

should I add any sensor in the yaml file?

Thank you very much!

erikwijmans, 2019-10-03:
Moved this to habitat-api so we can keep issues categorized

erikwijmans, 2019-10-12:
https://github.com/facebookresearch/habitat-api/blob/master/habitat/sims/habitat_simulator/habitat_simulator.py#L370 (https://aihabitat.org/docs/habitat-sim/habitat_sim.agent.AgentState.html) contains all of that information.  It doesn't look like we have that exposed directly through any sensors or measures.  Feel free to add your own.

chenwydj, 2019-10-12:
I found this code snippet is helpful: [https://github.com/facebookresearch/habitat-api/issues/109#issuecomment-499710082](https://github.com/facebookresearch/habitat-api/issues/109#issuecomment-499710082)

closed 2019-10-30 by chenwydj