facebookresearch/habitat-lab #226: Details about agent's trajectory in info["top_down_map"]["map"]

chenwydj, 2019-10-12:
## ❓ Questions and Help

This [line](https://github.com/facebookresearch/habitat-api/blob/master/habitat/utils/visualizations/maps.py#L41) tries to color the agent's trajectory.

It treats all values above (inclusive) 10 as the agent's trajectory. I dig into it and found top_down_map could have larger values like 11, 12, 24...

May I ask the detailed meaning of values above (inclusive) 10 in the top_down_map? (which can be obtained in info["top_down_map"]["map"] by turning on the "HEADING_SENSOR" and "TOP_DOWN_MAP" measurement)

Thank you!

erikwijmans, 2019-10-15:
All those values are here, with names: https://github.com/facebookresearch/habitat-api/blob/master/habitat/utils/visualizations/maps.py#L34

rpartsey, 2022-09-01:
> It treats all values above (inclusive) 10 as the agent's trajectory. I dig into it and found top_down_map could have larger values like 11, 12, 24...

`info["top_down_map"]["map"]` is a 2D array of indices representing a colors in the habitat.utils.visualizations.maps.TOP_DOWN_MAP_COLORS. The values >= 10 correspond to the agent's path trajectory color indices.

closed 2022-09-01 by rpartsey